micropython/micropython #6029: nrf.send() is there a limit ? (esp32)

JirkaVer, 2020-05-11:
I have a project on 2 esp32's with nr24l01. It works like a charm until I want to send 4 integers instead of 3 using nrf.send() 

I have following code on master side

```
import machine
import utime
#https://github.com/micropython/micropython/tree/master/drivers/nrf24l01
import sys
import ustruct as struct
import utime
from machine import Pin, SPI
from nrf24l01 import NRF24L01
from micropython import const

# Slave pause between receiving data and checking for further packets.
_RX_POLL_DELAY = const(15)
# Slave pauses an additional _SLAVE_SEND_DELAY ms after receiving data and before
# transmitting to allow the (remote) master time to get into receive mode. The
# master may be a slow device. Value tested with Pyboard, ESP32 and ESP8266.
_SLAVE_SEND_DELAY = const(10)
cfg = {"spi": -1, "miso": 19, "mosi": 23, "sck": 18, "csn": 5, "ce": 4}
#http://electroniqueamateur.blogspot.com/2019/12/communication-nrf24l01-avec-cartes.html?m=0

# Addresses are in little-endian format. They correspond to big-endian
# 0xf0f0f0f0e1, 0xf0f0f0f0d2
pipes = (b"\xe1\xf0\xf0\xf0\xf0", b"\xd2\xf0\xf0\xf0\xf0")

###############################################################################
# Parameters and global variables

# Pin definitions
sda_pin = machine.Pin(21)
scl_pin = machine.Pin(22)

# Create an I2C object out of our SDA and SCL pin objects
i2c = machine.I2C(sda=sda_pin, scl=scl_pin)

# TMP102 address on the I2C bus
tmp102_addr = 0x48

# TMP102 register addresses
reg_temp = 0x00
reg_config = 0x01

pakketnr = 0
###############################################################################
# Functions

# Read temperature registers and calculate Celsius
def read_temp():

    return 221;

def updatePakketnr():
  global pakketnr
  pakketnr +=  1
  
  if (pakketnr >= 3600):
    pakketnr = 0
  return pakketnr

# Initialize communications with the TMP102
def init():

    # Read CONFIG register (2 bytes) and convert to integer list
    val = i2c.readfrom_mem(tmp102_addr, reg_config, 2)
    val = list(val)

    # Set to 4 Hz sampling (CR1, CR0 = 0b10)
    val1 = val1 & 0b00111111
    val1 = val1 | (0b10 << 6)

    # Write 4 Hz sampling back to CONFIG
    i2c.writeto_mem(tmp102_addr, reg_config, bytearray(val))
    
    

def readSendTemp():
    csn = Pin(cfg["csn"], mode=Pin.OUT, value=1)
    ce = Pin(cfg["ce"], mode=Pin.OUT, value=0)
    if cfg["spi"] == -1:
        spi = SPI(-1, sck=Pin(cfg["sck"]), mosi=Pin(cfg["mosi"]), miso=Pin(cfg["miso"]))
        nrf = NRF24L01(spi, csn, ce, payload_size=12)
    else:
        nrf = NRF24L01(SPI(cfg["spi"]), csn, ce, payload_size=12)
    
    nrf.open_tx_pipe(pipes[0])

    nrf.open_rx_pipe(1, pipes[1])
    nrf.start_listening()
    
    print("NRF24L01 master mode")

    while True:
        # stop listening and send packet
        nrf.stop_listening()
        temperature = read_temp()
        preamble = 0xAA
        
        counterBit = 0
        for bit in bin(temperature):
          counterBit +=1
         
        counterBit -= 2 
        
        numberOfBytes = int(counterBit / 8)
        updatePakketnr()
        
        print("sending:",preamble,numberOfBytes,int(temperature*100))
               
        
        
        try:
            nrf.send(struct.pack("iiii",preamble,numberOfBytes,pakketnr,temperature))
            print("sent")
        except OSError:
            pass

       
        utime.sleep(5)

def readout():
  while True:
    # Read temperature and print it to the console

    temperature = read_temp()
    print(round(temperature, 2))
    utime.sleep(1)

readSendTemp()

```

and following on slave side

```
import machine
import sys
import utime

"""import machine
import sys
import utime

"""Test for nrf24l01 module.  Portable between MicroPython targets."""
#https://github.com/micropython/micropython/tree/master/drivers/nrf24l01
import sys
import ustruct as struct
import utime
from machine import Pin, SPI
from nrf24l01 import NRF24L01
from micropython import const
# Slave pause between receiving data and checking for further packets.
_RX_POLL_DELAY = const(15)
# Slave pauses an additional _SLAVE_SEND_DELAY ms after receiving data and before
# transmitting to allow the (remote) master time to get into receive mode. The
# master may be a slow device. Value tested with Pyboard, ESP32 and ESP8266.
_SLAVE_SEND_DELAY = const(10)

cfg = {"spi": -1, "miso": 19, "mosi": 23, "sck": 18, "csn": 5, "ce": 4}
 

# Addresses are in little-endian format. They correspond to big-endian
# 0xf0f0f0f0e1, 0xf0f0f0f0d2
pipes = (b"\xe1\xf0\xf0\xf0\xf0", b"\xd2\xf0\xf0\xf0\xf0")

###############################################################################
# Parameters and global variables

# Pin definitions
sda_pin = machine.Pin(21)
scl_pin = machine.Pin(22)

# Create an I2C object out of our SDA and SCL pin objects
i2c = machine.I2C(sda=sda_pin, scl=scl_pin)

# TMP102 address on the I2C bus
tmp102_addr = 0x48

# TMP102 register addresses
reg_temp = 0x00
reg_config = 0x01

def slave():
    csn = Pin(cfg["csn"], mode=Pin.OUT, value=1)
    ce = Pin(cfg["ce"], mode=Pin.OUT, value=0)
    if cfg["spi"] == -1:
        spi = SPI(-1, sck=Pin(cfg["sck"]), mosi=Pin(cfg["mosi"]), miso=Pin(cfg["miso"]))
        nrf = NRF24L01(spi, csn, ce, payload_size=12)
    else:
        nrf = NRF24L01(SPI(cfg["spi"]), csn, ce, payload_size=12)

    nrf.open_tx_pipe(pipes[1])
    nrf.open_rx_pipe(1, pipes[0])
    nrf.start_listening()

    print("NRF24L01 slave mode, waiting for packets(ctrl-C to stop)")

    while True:
        if nrf.any():
            while nrf.any():
                buf = nrf.recv()
                recieved = struct.unpack("iiii", buf) 
                
                preamble = recieved[0]
                numberOfBytes = recieved[1]
                temp = recieved[2]/100.0
                
                print("preamble",hex(preamble),"numberofbytes",numberOfBytes,"temperature", temp) #Dit wordt ingelezen in de COM AUB VERANDER DE STRUCTUUR NIET 
                utime.sleep_ms(_RX_POLL_DELAY)
            nrf.start_listening()

slave()
```

These are build from nrf24l01test.py.

Now I was wondering is there a limit in the send function ? Or is there something else going wrong?

I tried sending 4 constants too. But that doesn't seem to help either

Thanks a lot already

tve, 2020-05-11:
Please post this type of question on forum.micropython.org
(Sorry, I don't have an answer to your question)

JirkaVer, 2020-05-11:
> 
> 
> Please post this type of question on forum.micropython.org
> (Sorry, I don't have an answer to your question)

Thanks for the reply I posted it there too :D

closed 2020-05-11 by JirkaVer

JirkaVer, 2020-05-19:
Eventually I found it. My payload was too small.

This just in case someone has a similar problem in the future