micropython/micropython #6031: Building NUCLEO_H743ZI in windows

infantasteff, 2020-05-12:
Hello,

I am currently trying to build micropython in windows 10 in atollic true studio ide for port NUCLEO_H743ZI. I got the following error. 

c:/program files (x86)/gnu tools arm embedded/9 2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/thumb/v7e-m+dp/hard/libgcc.a
c:\Program Files (x86)\GNU Tools Arm Embedded\9 2019-q4-major\bin\arm-none-eabi-ld.exe: cannot find build-NUCLEO_H743ZI/lib/mbedtls/library/ssl_: No such file or directory
make: *** [Makefile:614: build-NUCLEO_H743ZI/firmware.elf] Error 1

dpgeorge, 2020-05-13:
Did you update the submodules?  Try running `make submodules` in the `ports/stm32/` directory.

infantasteff, 2020-05-13:
I updated the submodules again. Below is the list of submodules updated
make V=1 submodules
Updating submodules: lib/lwip lib/mbedtls lib/mynewt-nimble lib/stm32lib
git submodule update --init ../../lib/lwip ../../lib/mbedtls ../../lib/mynewt-nimble ../../lib/stm32lib

But still I receive the same error message

dpgeorge, 2020-05-13:
Please copy here the full output of `make V=1`

infantasteff, 2020-05-13:
I have attached the full output for make V=1
[micropython3.build.log](https://github.com/micropython/micropython/files/4620950/micropython3.build.log)

stinos, 2020-05-13:
Command line length problem: the `srv` in `ssl_srv` starts at column 16385.
A bit strange, because the only limits I know of are around 260, 2048, 8192 and 32768 depending on how a command gets invoked. Don't think I've seen 16384 before, and I don't know which component is at fault here.

stinos, 2020-05-13:
Workaround: change build directory to `b` instead of `build-NUCLEO_H743ZI`.

infantasteff, 2020-05-13:
I am able to generate the binaries. 
Thank you very much

closed 2020-05-13 by infantasteff