micropython/micropython #6032: Error on nativeglue.h while trying to build new firmware

arlucio, 2020-05-12:
When I tried to make this file [Makefile](https://github.com/pulkin/micropython/blob/master/ports/gprs_a9/Makefile) to update micropython version of this port.

In file included from ../../py/emitnx86.c:4:
../../py/nativeglue.h:138: error: expected ‘)’ before ‘->’ token
../../py/nativeglue.h:140: error: expected ‘;’ before ‘void’

Anyone knows what could it be? So I may to have a hint where to search for a solution.

dlech, 2020-05-12:
Perhaps a macro is not properly defined? I think there is a compiler option to save the output of a file after running the preprocessor. That could be useful to debug the problem.

dpgeorge, 2020-05-13:
It looks like it's because an entry in `mp_fun_table_t` is called `setjmp` and probably the toolchain redefines that symbol to something else.

A fix would be to rename that entry (line 138) from `setjmp` to `setjmp_`.

arlucio, 2020-05-13:
Thanks for the quick response. After I renamed that entry the error disappeared.

I'm getting a new error now so I've opened a new issue at #6035.

closed 2020-05-13 by arlucio

dpgeorge, 2020-05-14:
Fixed by 801f7dca78c2983a8ab96e79784d0b3c5cc9e7de