micropython/micropython #6034: Running script via pyboard.py not working with mimxrt port

alphaFred, 2020-05-13:
I figured out that it seems like this behavior is dependent on the amount of bytes you send to the board running micropython. That is the amount of code in the script you want to run (including white space of course). As soon as pyboard.py transmits more then 64bytes an error occurs.
This error results in pyboard.py freezing and the script execution not taking place.

When I hook up my debugger I can see that I run into an USB error:
[image]

This is the  respective call stack:
[image]

I have to admit that I am unfortunately not well versed with USB. According to the datasheet of the controller this error flag indicates the following:

> When completion of a USB transaction results in an error condition, this bit is set by the Host/Device Controller.

Maybe someone else has faced this issue and could give me a hint for debugging or has any idea what exactly this "USB error" could be.

dpgeorge, 2020-05-14:
I would suggest building and running one of the mimxrt examples from TinyUSB to see if it has a similar issue.  That will help narrow down the issue.

dpgeorge, 2020-05-15:
> I would suggest building and running one of the mimxrt examples from TinyUSB to see if it has a similar issue.

I just tried building the TinyUSB `cdc_msc` example for the `teensy_40` board at TinyUSB current master https://github.com/hathach/tinyusb/commit/be9f938bd33d26d07d1bd1ff479e22f440777a63 .  And it seems to have the same problem with >64 bytes sent at once.  Tested by cat'ing a file to the USB serial port, 64 byte file works fine, 65 byte file makes the board reset.

alphaFred, 2020-05-15:
Thanks for the update. Unfortunately I will only find some time over the weekend. 

I will go ahead testing a cdc example project from NXP.  Maybe I can find some differences in the configuration.

What puzzles me though is the reset which you mentioned. In my call stack I saw the `Reset Exception`, too. Maybe we have a buffer overflow 🤔.

dpgeorge, 2020-05-15:
Note that 64 bytes is the maximum that will fit in a USB transfer for the CDC configuration.  Eg on the stm32 port there is a special case to handle 64 byte dev->host transfers (see `usbd_cdc_tx_ready`).  But the fact TinyUSB works ok on other MCUs points to a low-level issue in the mimxrt USB code.

hathach, 2020-05-28:
> I just tried building the TinyUSB `cdc_msc` example for the `teensy_40` board at TinyUSB current master [hathach/tinyusb@be9f938](https://github.com/hathach/tinyusb/commit/be9f938bd33d26d07d1bd1ff479e22f440777a63) . And it seems to have the same problem with >64 bytes sent at once. Tested by cat'ing a file to the USB serial port, 64 byte file works fine, 65 byte file makes the board reset.

ola, @alphaFred @dpgeorge it sounds like a tinyusb bug, would you mind opening an issue there. I will try to help with the troubleshooting. Thanks

dpgeorge, 2020-05-28:
Bug report filed at TinyUSB.

alphaFred, 2020-06-07:
Hi @dpgeorge,
I have tested the suggested fix from hathach ([my comment with more details on tinyusb issue #423](https://github.com/hathach/tinyusb/issues/423#issuecomment-640178534)) - would you say we should use that as a preliminary fix for the mimxrt port?

This would allow to set up automatic testing with pyboard.py as @jimmo has commented [here](https://github.com/micropython/micropython/pull/5919#issuecomment-616287767).

dpgeorge, 2020-06-07:
@alphaFred I've now also tested your fix on a Teensy 4.0 and it works well, and it allowed me to start running the uPy test suite on this board.  The tests have random failures (like the same test run again and again will eventually fail, eg `basics/fun_callstar.py`) but I didn't yet try to investigate the problem with that.  It could still be a USB issue, or something else.

While it would be good to get a proper fix from upstream TinyUSB, I think it's worth applying your fix now to this repo because it drastically improves usability of USB on this port, and allows to run other tests.  If you could open a PR with your fix that'd be great.

alphaFred, 2020-06-07:
> @alphaFred I've now also tested your fix on a Teensy 4.0 and it works well, and it allowed me to start running the uPy test suite on this board.  The tests have random failures (like the same test [...]

Sure. I will create one today.

dpgeorge, 2020-06-08:
Preliminary fix for this issue is now done by 6ac05af8e1408a1f5cd27d57c8581a9f4197a2f5

closed 2020-06-08 by dpgeorge

dpgeorge, 2020-07-19:
This was fixed properly in TinyUSB in https://github.com/hathach/tinyusb/pull/464 .  Might be worth seeing if any changes from there can be applied here.

alphaFred, 2020-07-20:
Thanks for the info. Already got notified from tinyUSB project 😄. 

I will take a look.

alphaFred, 2020-07-27:
@dpgeorge Initial testing of tinyUSB changes are looking good. In order to take this over I will need to make changes to all platform ports due to interface changes. This will take some more time. I hope that the CI will catch any issues 🙈

hathach, 2020-07-27:
@alphaFred If you don't plan to have micropython running with FS host such as other MCUs. You can actually get away without the FullSpeed configuration descriptor. Since all the PC should be able to do HighSpeed. I mean you can do it gradually later on :)