micropython/micropython #6035: Error on py/stream.c while trying to build new firmware (errno related)

arlucio, 2020-05-13:
When I tried to build this file [Makefile](https://github.com/pulkin/micropython/blob/master/ports/gprs_a9/Makefile) to update micropython version of the related port. I'm getting this error:

```
CC ../../py/stream.c
../../py/stream.c: In function ‘mp_stream_posix_write’:
../../py/stream.c:511: error: lvalue required as unary ‘&’ operand
../../py/stream.c: In function ‘mp_stream_posix_read’:
../../py/stream.c:522: error: lvalue required as unary ‘&’ operand
../../py/stream.c: In function ‘mp_stream_posix_lseek’:
../../py/stream.c:536: error: lvalue required as unary ‘&’ operand
../../py/stream.c: In function ‘mp_stream_posix_fsync’:
../../py/stream.c:546: error: lvalue required as unary ‘&’ operand
```

I  believe it is related to #5959, so I tried adding `embed/__errno.c \`  to `LIB_SRC_C = $(addprefix lib/,\` on the Makefile, and tried to build it again. The error persisted (actually that makes sense as the port has an own errno.h [here](https://github.com/pulkin/GPRS_C_SDK/blob/982fd50735f5efb1ee15ed1bdbdcbcc4e5f33c5b/include/std_inc/errno.h) )

I changed value of MICROPY_STREAMS_POSIX_API from 1 to 0 on mpconfigport.h and got no more of this error but then got:

```
CC ../../lib/axtls/ssl/asn1.c
In file included from ../../lib/axtls/ssl/asn1.c:39:
../../lib/axtls/ssl/os_port.h:51: error: expected ‘)’ before ‘->’ token
```
and commenting the line give future problems when trying to build the axtls lib, like:

```
build/lib/axtls/ssl/tls1.o: In function `send_raw_packet':
/home/deko/dev/full_micropython_for_search/micropython/ports/gprs_a9/../../lib/axtls/ssl/tls1.c:1062: undefined reference to `mp_stream_posix_write'
```
So seems like I need the `MICROPY_STREAMS_POSIX_API` set to 1 to build it, and I guess i would need to fix somehow the errno import?

Any hints on how should I proceed in this matter?

dpgeorge, 2020-05-14:
> I believe it is related to #5959,

Yes it because of that PR, specifically commit e08ca78f40bb1948acffa60c0315083acfb02227

> ./../py/stream.c:511: error: lvalue required as unary ‘&’ operand

The man page for errno states that "errno is defined by the ISO C standard to be a modifiable lvalue of type int".  It looks like the system you are building for defines `errno` to something which can't have its address taken, so is not ISO C compatible.

Indeed it look like it uses:
```c
#define errno                CSDK_FUNC(Socket_GetLastError)()
...
struct vtable {
 int                 (*Socket_GetLastError)(void);
}
```
This won't work because `Socket_GetLastError` returns an int, rather than a pointer to an int, so it's impossible to modify the errno value (as required by ISO C).

It looks like there was an attempt to fix this in `ram_pointers.h`:
```c
#define errno (*((volatile int *) 0x820a0c14))
```
but that previous definition is taking preference.

arlucio, 2020-05-15:
Thanks for the response! I commented the line and the error disappeared.

closed 2020-05-15 by arlucio