micropython/micropython #6036: ESP32 WiFi AP Channel Width

gigawatts, 2020-05-13:
It seems when creating a WiFi AP, it defaults to a 40 MHz channel bandwidth.

```
import network
ap = network.WLAN(network.AP_IF)
ap.active(True)
ap.config(essid="TEST")
ap.config(authmode=0, channel=2)
```

In the above example, when viewing the broadcasted SSID with an Android app such as "Wifi Analyzer", shows "CH 2+6" and "2447 - 2407 = 40 MHz".

Is there a setting that can be changed to set this to 20 MHz, so it only uses one channel?

tve, 2020-05-14:
I can confirm that I see the same. Looks like [esp_wifi_set_bandwidth](https://docs.espressif.com/projects/esp-idf/en/v4.0/api-reference/network/esp_wifi.html#_CPPv422esp_wifi_set_bandwidth16wifi_interface_t16wifi_bandwidth_t) is the solution. Looks like a bandwidth option should be added to the `config` method.

dpgeorge, 2020-05-14:
> Looks like a bandwidth option should be added to the `config` method.

+1

davefes, 2024-05-27:
+1  I actually see 60MHz on `MicroPython v1.22.2 on 2024-02-22; Generic ESP32 module with SPIRAM with ESP32` and the channel can not be set.