micropython/micropython #6037: ESP32: VfsLfs2, not auto start main.py if not in root. (uos.chdir)

straga, 2020-05-14:
Data Partition: VfsLfs2 

[image]

boot.py
``` python
part_name = "factory"
sys.path.append("/{}".format(part_name))
sys.path.append("/{}/{}".format(part_name, "lib"))
sys.path.append("/{}/{}".format(part_name, "app"))
uos.chdir("/{}".format(part_name))
```
Boot Board. Not Auto Start
[image]

Run by Hand: 
``` python
import main
main.main()
```

[image]

With vFat works and runs automatically.

dpgeorge, 2020-05-14:
Thanks for the report.  This is similar to #5981 but this time it's the `import_stat` function which does not take into account the current working directory.

dpgeorge, 2020-05-14:
See #6040 for a fix.

straga, 2020-05-14:
> See #6040 for a fix.
with that fix, now autorun main.py works.
Another thing works not stable and gets guru meditation: if a lot with read/write files.
Same project with vFat works like a charm.

dpgeorge, 2020-05-14:
> Another thing works not stable and gets guru meditation: if a lot with read/write files.

Can you provide code which shows the problem?  Or at least the crash output with a backtrace or the corresponding .elf file?

straga, 2020-05-15:
VfsLfs:
[image]

[bakctrace.txt](https://github.com/micropython/micropython/files/4633247/bakctrace.txt)
[application.zip](https://github.com/micropython/micropython/files/4633253/application.zip)

If it not help, i can provide full code with step by step.

dpgeorge, 2020-05-15:
Thanks for the trace.  I can see that the code that crashes is running in a separate thread, and it uses a context manager for the open file, and it crashes when trying to close the file exiting the context manager.

What exact version/commit of MicroPython do you use?  Is it modified?  What IDF version and what hardware do you run on?  Does it use PSRAM?

It might be because of thread stack overflow.  Did you use `_thread.stack_size()` to change the default?  The default might not be enough, you can try executing `_thread.stack_size(8*1024)` before any `start_new_thread()` and see if that helps.

straga, 2020-05-15:
Yep. You right.
I am using ESP32 without PSRAM for that test.
It was: 4 * 1024, I am decrise size, when was a lot module loaded. It help on board with out PSRAM.
Hm, for vFat it enough, but with vLfs2 not.
Now I am change to default 8 * 1024 - now works fine.
``` python
    loop = asyncio.get_event_loop()
    _ = _thread.stack_size(8 * 1024)
    _thread.start_new_thread(loop.run_forever, ())
```
Thanks a lot.

dpgeorge, 2020-05-16:
> Hm, for vFat it enough, but with vLfs2 not.
> Now I am change to default 8 * 1024 - now works fine.

Ok.  It was probably just on the edge with FAT (and may have had silent issues that didn't lead to a crash).  For LFS2 you can try 5*1024 or 6*1024, that's probably enough.

I'll close this issue because the original problem of starting main.py is now fixed by 7dffbfd22a128d88cd4cd7f23ebf24c46ae8791f

closed 2020-05-16 by dpgeorge