micropython/micropython #6042: SD card cannot be mounted on ESP-WROOM-32 using SDMMC.

UmFqYXQ, 2020-05-14:
>>> import uos
>>> import machine
>>> uos.mount(machine.SDCard(width=4), '/sd')
[0;32mI (59190) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;31mE (59220) sdmmc_sd: sdmmc_init_sd_scr: send_scr (1) returned 0x109[0m
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 16

I am using MicroPython firmware on ESP-WROOM-32. I have successfully mounted SD card using the SPI interface but I am unable to do the same using SDMMC. I am on Windows 10 using ESPlorer IDE.

Circuit connections :-
[image]
[Additionally :-
I don't know why but whenever I do a hard reset micropython repl never shows up.
It keeps on showing something like this:-
rst:0x10 (RTCWDT_RTC_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4844
load:0x40078000,len:9656
load:0x40080400,len:6216
csum err:0x70!=0xde
ets_main.c 371 
ets Jun  8 2016 00:22:57
The csum err: keeps on changing.
I figured out that this is because pin 12 is pulled up.
So for avoiding this I did the following:-
[image]
]

Output :-
[image]

(Please bare with my poor English.)

UmFqYXQ, 2020-05-14:
The connections are as per https://docs.micropython.org/en/latest/library/machine.SDCard.html#esp32 and 
[image]

UmFqYXQ, 2020-05-14:
Error Codes: https://github.com/micropython/micropython-lib/blob/master/errno/errno.py

miketeachman, 2020-05-15:
Two suggestions:
1. configure the SC Card in 1-line mode which does not use GPIO12.  In that case, you will not need a pull-up on GPIO12.
`uos.mount(machine.SDCard(width=1), '/sd')`
2. switch to a WROVER module which requires GPIO12 to be pulled high at reset.  e.g. the pullup you add to GPIO12 does not affect flash voltage selection at boot.

This issue is not a MicroPython bug or deficiency.  It is hardware related.  My recommendation is to close this issue and start a discussion on the MicroPython forum or maybe search the ESP32 forum for a hardware solution.

UmFqYXQ, 2020-05-15:
Thanks for the reply.
How can I configure the SD Card in one line mode?
import uos
import machine
uos.mount(machine.SDCard(), '/sd')
and disconnecting GPIO 4, 12 and 13 (and respective pull-ups) from SD Card would do it?
(Just making sure.)

UmFqYXQ, 2020-05-15:
When I want to hard reset the ESP32 I hold the push button connected to GPIO 12 pressed (this will pull-down GPIO 12), then I will press the reset button and after some seconds I will release the push button connected to GPIO 12. Thus I have no problems at the time of booting and GPIO 12 is now pulled-up again.

Is there anything wrong here which can cause problems in configuring the SD Card in 4-line mode?

UmFqYXQ, 2020-05-15:
>This issue is not a MicroPython bug or deficiency. It is hardware related. My recommendation is to >close this issue and start a discussion on the MicroPython forum or maybe search the ESP32 forum >for a hardware solution.
I would like to close this issue and start a discussion on micropython forum, but before that please tell me the answer of this question:-
"On the forum main page you can see several micropython boards in the background of the header image. What is the name of the I/O port between B9 and C1?: "
This question is being asked on MicroPython Register page (as you might know) and I because am new to micropython and programming microcontrollers I don't know the answer of this question.
This question is only the barrier for me to start a discussion for finding a solution to my issue so please consider helping me.

UmFqYXQ, 2020-05-15:
"""
This issue is not a MicroPython bug or deficiency. It is hardware related. My recommendation is to close this issue and start a discussion on the MicroPython forum or maybe search the ESP32 forum for a hardware solution.
"""
I would like to close this issue and start a discussion on micropython forum, but before that please tell me the answer of this question:-
"On the forum main page you can see several micropython boards in the background of the header image. What is the name of the I/O port between B9 and C1?: "

This question is being asked on MicroPython Register page (as you might know) and because I am new to micropython and programming microcontrollers I don't know the answer of this question.
This question is only the barrier for me to start a discussion for finding a solution to my issue so please consider helping me.

UmFqYXQ, 2020-05-15:
"""
This issue is not a MicroPython bug or deficiency. It is hardware related. My recommendation is to close this issue and start a discussion on the MicroPython forum or maybe search the ESP32 forum for a hardware solution.
"""
I would like to close this issue and start a discussion on micropython forum, but before that please tell me the answer of this question:-
"On the forum main page you can see several micropython boards in the background of the header image. What is the name of the I/O port between B9 and C1?: "

This question is being asked on MicroPython Register page (as you might know) and because I am new to micropython and programming microcontrollers I don't know the answer of this question.
This question is only the barrier for me to start a discussion for finding a solution to my issue so please consider helping me.

Thank you for your help.

tve, 2020-05-15:
> I figured out that this is because pin 12 is pulled up.

Where? Have you tried using the internal pull-up?

Connecting a 10uF cap to gpio12 guarantees that SDMMC will never work! No cap.

miketeachman, 2020-05-15:
> What is the name of the I/O port between B9 and C1?

C0

You will get lots of help on the Forum.

UmFqYXQ, 2020-05-16:
> > I figured out that this is because pin 12 is pulled up.
> 
> Where? Have you tried using the internal pull-up?

No, as GPIO 2, 4, 12, 13, and 14 are being used for interfacing SD Card (in 4-line mode), I temporarily removed the jumper and pull-up resistor connected to only one GPIO (at a time) among the mentioned and pressed the reset button. I repeated the same procedure for all GPIOs and found that when GPIO 12 was pulled-up the repl was not showing up. I then disconnected the jumper and pull-up resistor connected to GPIO 12 and measured the voltage level it was zero during hard resetting the board.

> Connecting a 10uF cap to gpio12 guarantees that SDMMC will never work! No cap.

Can you please tell tell why?
I just connected the capacitor just for debouncing the push button (simplest debouncing circuit). Am I doing something wrong here?

UmFqYXQ, 2020-05-16:
> 
> 
> > What is the name of the I/O port between B9 and C1?
> 
> C0
> 
> You will get lots of help on the Forum.

Thank you very much for your help.
I got the answer, it was a very silly question!

UmFqYXQ, 2020-05-16:
> 
> 
> > What is the name of the I/O port between B9 and C1?
> 
> C0
> 
> You will get lots of help on the Forum.

Thank you very much for your help.
I thought that it is just a I/O pin and not a port. I got the answer, it was a very silly question!

closed 2020-05-17 by UmFqYXQ