micropython/micropython #6045: Memory leak for native machine code in mpy files

pmvr, 2020-05-15:
I noticed a memory leak in the usage of native mpy files.

Even for the simple example given in the [documentation](https://docs.micropython.org/en/latest/develop/natmod.html#natmod) the following code demonstrates the issue:

```
import factorial
from urandom import randint
import gc

for i in range(100000):
    c = factorial.factorial(randint(4, 15))
    print(gc.mem_free())
    print(i, c)

```

I tested this with
> MicroPython v1.12 on 2019-12-20; PYBD-SF2W with STM32F722IEK

on a Pyboad D-series.

dpgeorge, 2020-05-16:
If you change `randint(4, 15)` to `randint(4, 12)` then the "memory leak" is not there.

What's happening is that for large factorials the value computed is too large to be stored in a small-int (which takes 0 heap memory) and instead a big-int is created (on the heap).  The function `mp_obj_new_int(...)` is doing the memory allocation in this case, for the large values.

And it's not a memory leak because a `gc.collect()` will reclaim it when the number is no longer referenced.

pmvr, 2020-05-16:
Right, a `gc.collect()` reclaims the memory.

However, there still seems to be a difference:
The loop over `factorial.factorial(15)` lets the board run out of memory until it resets, the same loop over the simple python function `f(15)`
```
def f(n):
    r = 1
    for i in range(2, n+1):
        r *= i
    return r
```
lets the garbage collector release memory right before the board resets.

One further remark: `factorial.factorial(15)` in fact computes 15! % 2**32, because `factorial_helper` operates on `mp_int_t`, which is on STM32: `preprocessed.h:typedef int mp_int_t;` So, for a 31-bit integer, a big-int is created on the heap.

pmvr, 2020-05-16:
Overall, I think, its clear enough right now. Thanks!

dpgeorge, 2020-05-21:
> The loop over `factorial.factorial(15)` lets the board run out of memory until it resets,

Yes, I tested this and it does do a hard reset after a GC with the native code.

This is actually a bug: on stm32 the native code loaded from a .mpy file is not reachable by the GC because there is no pointer stored which points to the start of the native chunk of code.

dpgeorge, 2020-05-21:
See #6061 for a proposed fix.

pmvr, 2020-05-24:
Yes, I can confirm that the proposed fix lets the GC automatically release unused memory from the heap of a native mpy-module.

commit 9883d8e8 (2020-08-02): py/persistentcode: Maintain root ptr list of imported native .mpy code.

On ports where normal heap memory can contain executable code (eg ARM-based
ports such as stm32), native code loaded from an .mpy file may be reclaimed
by the GC because there's no reference to the very start of the native
machine code block that is reachable from root pointers (only pointers to
internal parts of the machine code block are reachable, but that doesn't
help the GC find the memory).

This commit fixes this issue by maintaining an explicit list of root
pointers pointing to native code that is loaded from an . [... cut]
PR #6290 (2020-08-02): py/persistentcode: Maintain root ptr list of imported native .mpy code.
(This is an alternative to #6061 that is more comprehensive and robust.)

On ports where normal heap memory can contain executable code (eg ARM-based ports such as stm32), native code loaded from an .mpy file may be reclaimed by the GC because there's no reference to the very start of the native machine code block that is reachable from root pointers (only pointers to internal parts of the machine code block are reachable, but that doesn't help the GC find the memory).

This commit fixes this issue by maintaining an explic [... cut]
closed 2020-08-02 by dpgeorge