micropython/micropython #6046: Cannot build any stm32 targets with gcc-9 or gcc-10

alexxy, 2020-05-15:
Hi!

I'm trying to build micropython for stm32f4 target (actualy tryed many targets to test) with arm-none-eabi toolchain based on gcc-9 or gcc-10. I get following error

```
arm-none-eabi-gcc -I. -I../.. -Ibuild-PYBV11 -I../../lib/cmsis/inc -I../../lib/stm32lib/CMSIS/STM32F4xx/Include/ -I../../lib/stm32lib/STM32F4xx_HAL_Driver/Inc -Iusbdev/core/inc -Iusbdev/class/inc -Ilwip_inc -Wall -Wpointer-arith -Werror -Wdouble-promotion -Wfloat-conversion -std=gnu99 -nostdlib -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -DMICROPY_VFS_FAT=1 -DMICROPY_VFS_LFS2=1 -DLFS2_NO_MALLOC -DLFS2_NO_DEBUG -DLFS2_NO_WARN -DLFS2_NO_ERROR -DLFS2_NO_ASSERT  -DSTM32F405xx -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mtune=cortex-m4 -mcpu=cortex-m4 -Os -DNDEBUG -O2 -Iboards/PYBV11 -DSTM32_HAL_H='<stm32f4xx_hal.h>' -DMICROPY_HW_VTOR=0x08000000 -DMICROPY_FLOAT_IMPL=MICROPY_FLOAT_IMPL_FLOAT -fsingle-precision-constant -fdata-sections -ffunction-sections -DMICROPY_MODULE_FROZEN_STR -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool -DMICROPY_MODULE_FROZEN_MPY -DMICROPY_ROM_TEXT_COMPRESSION=1 -c -MD -o build-PYBV11/lib/libc/string0.o ../../lib/libc/string0.c -lc -lnosys
In file included from /usr/arm-none-eabi/include/ssp/strings.h:34,
                 from /usr/arm-none-eabi/include/strings.h:77,
                 from /usr/arm-none-eabi/include/string.h:24,
                 from ../../lib/libc/string0.c:28:
../../lib/libc/string0.c:32:7: error: expected declaration specifiers or '...' before numeric constant
   32 | void *memcpy(void *dst, const void *src, size_t n) {
      |       ^~~~~~
In file included from /usr/arm-none-eabi/include/string.h:180,
                 from ../../lib/libc/string0.c:28:
../../lib/libc/string0.c:32:7: error: expected ')' before '!=' token
   32 | void *memcpy(void *dst, const void *src, size_t n) {
      |       ^~~~~~
../../lib/libc/string0.c:32:7: error: expected ')' before '?' token
   32 | void *memcpy(void *dst, const void *src, size_t n) {
      |       ^~~~~~
In file included from /usr/arm-none-eabi/include/ssp/strings.h:34,
                 from /usr/arm-none-eabi/include/strings.h:77,
                 from /usr/arm-none-eabi/include/string.h:24,
                 from ../../lib/libc/string0.c:28:
../../lib/libc/string0.c:67:7: error: expected declaration specifiers or '...' before numeric constant
   67 | void *memmove(void *dest, const void *src, size_t n) {
      |       ^~~~~~~
In file included from /usr/arm-none-eabi/include/string.h:180,
                 from ../../lib/libc/string0.c:28:
../../lib/libc/string0.c:67:7: error: expected ')' before '!=' token
   67 | void *memmove(void *dest, const void *src, size_t n) {
      |       ^~~~~~~
../../lib/libc/string0.c:67:7: error: expected ')' before '?' token
   67 | void *memmove(void *dest, const void *src, size_t n) {
      |       ^~~~~~~
In file included from /usr/arm-none-eabi/include/ssp/strings.h:34,
                 from /usr/arm-none-eabi/include/strings.h:77,
                 from /usr/arm-none-eabi/include/string.h:24,
                 from ../../lib/libc/string0.c:28:
../../lib/libc/string0.c:82:7: error: expected declaration specifiers or '...' before numeric constant
   82 | void *memset(void *s, int c, size_t n) {
      |       ^~~~~~
In file included from /usr/arm-none-eabi/include/string.h:180,
                 from ../../lib/libc/string0.c:28:
../../lib/libc/string0.c:82:7: error: expected ')' before '!=' token
   82 | void *memset(void *s, int c, size_t n) {
      |       ^~~~~~
../../lib/libc/string0.c:82:7: error: expected ')' before '?' token
   82 | void *memset(void *s, int c, size_t n) {
      |       ^~~~~~
In file included from /usr/arm-none-eabi/include/ssp/strings.h:34,
                 from /usr/arm-none-eabi/include/strings.h:77,
                 from /usr/arm-none-eabi/include/string.h:24,
                 from ../../lib/libc/string0.c:28:
../../lib/libc/string0.c:163:7: error: expected declaration specifiers or '...' before numeric constant
  163 | char *strcpy(char *dest, const char *src) {
      |       ^~~~~~
In file included from /usr/arm-none-eabi/include/string.h:180,
                 from ../../lib/libc/string0.c:28:
../../lib/libc/string0.c:163:7: error: expected ')' before '!=' token
  163 | char *strcpy(char *dest, const char *src) {
      |       ^~~~~~
../../lib/libc/string0.c:163:7: error: expected ')' before '?' token
  163 | char *strcpy(char *dest, const char *src) {
      |       ^~~~~~
In file included from /usr/arm-none-eabi/include/ssp/strings.h:34,
                 from /usr/arm-none-eabi/include/strings.h:77,
                 from /usr/arm-none-eabi/include/string.h:24,
                 from ../../lib/libc/string0.c:28:
../../lib/libc/string0.c:173:7: error: expected declaration specifiers or '...' before numeric constant
  173 | char *stpcpy(char *dest, const char *src) {
      |       ^~~~~~
In file included from /usr/arm-none-eabi/include/string.h:180,
                 from ../../lib/libc/string0.c:28:
../../lib/libc/string0.c:173:7: error: expected ')' before '!=' token
  173 | char *stpcpy(char *dest, const char *src) {
      |       ^~~~~~
../../lib/libc/string0.c:173:7: error: expected ')' before '?' token
  173 | char *stpcpy(char *dest, const char *src) {
      |       ^~~~~~
In file included from /usr/arm-none-eabi/include/ssp/strings.h:34,
                 from /usr/arm-none-eabi/include/strings.h:77,
                 from /usr/arm-none-eabi/include/string.h:24,
                 from ../../lib/libc/string0.c:28:
../../lib/libc/string0.c:181:7: error: expected declaration specifiers or '...' before numeric constant
  181 | char *strcat(char *dest, const char *src) {
      |       ^~~~~~
In file included from /usr/arm-none-eabi/include/string.h:180,
                 from ../../lib/libc/string0.c:28:
../../lib/libc/string0.c:181:7: error: expected ')' before '!=' token
  181 | char *strcat(char *dest, const char *src) {
      |       ^~~~~~
../../lib/libc/string0.c:181:7: error: expected ')' before '?' token
  181 | char *strcat(char *dest, const char *src) {
      |       ^~~~~~

```

dpgeorge, 2020-05-16:
gcc 9 and gcc 10 work ok for me, including for stm32 boards (using arm-none-eabi-xxx on Arch Linux).

The above looks like an issue with the C library header files (or something), redefining the string functions.

What is the contents of `/usr/arm-none-eabi/include/ssp/strings.h`?

alexxy, 2020-05-16:
This file comes from newlib toolchain. And i have no problems building apps with libopencm3

alexxy, 2020-05-16:
Also redefinition of memcpy and friends comes from string0.c file.

dpgeorge, 2020-05-16:
It could be that your toolchain enables `_FORTIFY_SOURCE` by default, which is not compatible with MicroPython.  To disable it try:
```
$ make CFLAGS_EXTRA="-U_FORTIFY_SOURCE"
```

alexxy, 2020-05-16:
Fortyfy source for embedded toolchain? And why in that case it works for libopencm3? 

Problem comes because string.h included in string0.c and it actualy uses toolchain provided string.h

alexxy, 2020-05-16:
> It could be that your toolchain enables `_FORTIFY_SOURCE` by default, which is not compatible with MicroPython. To disable it try:
> 
> ```
> $ make CFLAGS_EXTRA="-U_FORTIFY_SOURCE"
> ```

I checked this options. And seems like it builds now. So possible fix is to enable 
```
-U_FORTIFY_SOURCE 
```

by default

dpgeorge, 2021-11-17:
Fixed by 5cf71b55960651fc506df0ac41490f12dd1d63fd

closed 2021-11-17 by dpgeorge