micropython/micropython #6053: gcc 10 optimises memset to call itself

dpgeorge, 2020-05-17:
Using gcc 10.1.0 (tested with arm-none-eabi-gcc), the code in `lib/libc/string0.c` turns out to be optimised more than in previous versions of gcc.  The compiler now optimises the `memset` implementation it terms of itself (!).

Not sure exactly the best way to deal with this, but a current workaround is to add the following line to the Makefile:
```
$(BUILD)/lib/libc/string0.o: CFLAGS += -fno-builtin
```

stinos, 2020-05-17:
Do you mean `memset` gets replaced with a version recursively calling `memset`, and is this a gcc bug then?

dpgeorge, 2020-05-18:
> Do you mean `memset` gets replaced with a version recursively calling `memset`

Yes.  Part of the `memset` implementation recursively calls itself, leading to infinite recursion.

I wouldn't call it a bug in gcc.  gcc says it will potentially optimise any known built-in function, and that can mean replacing a call to (eg) `memset` with the equivalent inline code.  Or if it sees some code that's doing the equivalent of `memset` then it can replace that with a call.

The option `-ffreestanding` is also relevant here.

dpgeorge, 2020-05-21:
See #6060 for a proposed fix.

dpgeorge, 2020-05-27:
Fixed by a902b69dd51de0e3fe3bb6955296591d6a93abab

closed 2020-05-27 by dpgeorge

jcw, 2020-05-29:
I seem to be running into the same issue as of today, after latest git pull on unmodified clone.

Context:

* MacOS 10.15.5, build of ports/minimal/
* gcc -m32 -I. -I../.. -Ibuild -Wall -Werror -Wdouble-promotion -Wfloat-conversion -std=c99  -Os -DNDEBUG -fdata-sections -ffunction-sections -DMICROPY_ROM_TEXT_COMPRESSION=1 -ffreestanding -fno-builtin -fno-lto -c -MD -o build/lib/libc/string0.o ../../lib/libc/string0.c

```
$ gcc -v
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
Apple clang version 11.0.3 (clang-1103.0.32.62)
Target: x86_64-apple-darwin19.5.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
```

First few lines of compiler output:

```
../../lib/libc/string0.c:32:7: error: expected parameter declarator
void *memcpy(void *dst, const void *src, size_t n) {
      ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/secure/_string.h:63:46: note:
      expanded from macro 'memcpy'
                __builtin___memcpy_chk (dest, __VA_ARGS__, __darwin_obsz0 (dest))
                                                           ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/secure/_common.h:38:63: note:
      expanded from macro '__darwin_obsz0'
#define __darwin_obsz0(object) __builtin_object_size (object, 0)
                                                              ^
../../lib/libc/string0.c:32:7: error: expected ')'
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/secure/_string.h:63:46: note:
      expanded from macro 'memcpy'
                __builtin___memcpy_chk (dest, __VA_ARGS__, __darwin_obsz0 (dest))
[...]
```

dpgeorge, 2020-05-29:
@jcw it could be similar to #6046, which is fixed by `make CFLAGS_EXTRA="-U_FORTIFY_SOURCE"`

jcw, 2020-05-29:
I tried that, doesn't appear to work in my case:

```
$ make CFLAGS_EXTRA="-U_FORTIFY_SOURCE" V=1
python3 ../../py/makeversionhdr.py build/genhdr/mpversion.h
CC ../../lib/libc/string0.c
gcc -m32 -I. -I../.. -Ibuild -Wall -Werror -Wdouble-promotion -Wfloat-conversion -std=c99  -Os -DNDEBUG -fdata-sections -ffunction-sections -DMICROPY_ROM_TEXT_COMPRESSION=1 -ffreestanding -fno-builtin -fno-lto -c -MD -o build/lib/libc/string0.o ../../lib/libc/string0.c
../../lib/libc/string0.c:32:7: error: expected parameter declarator
void *memcpy(void *dst, const void *src, size_t n) {
      ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/secure/_string.h:63:46: note:
      expanded from macro 'memcpy'
                __builtin___memcpy_chk (dest, __VA_ARGS__, __darwin_obsz0 (dest))
```

Hm I don't see the flag getting added, trying manually - no, change:

```
$ gcc -m32 -I. -I../.. -Ibuild -Wall -Werror -Wdouble-promotion -Wfloat-conversion -std=c99  -Os -DNDEBUG -fdata-sections -ffunction-sections -DMICROPY_ROM_TEXT_COMPRESSION=1 -ffreestanding -fno-builtin -fno-lto -c -MD -o build/lib/libc/string0.o -U_FORTIFY_SOURCE ../../lib/libc/string0.c
../../lib/libc/string0.c:32:7: error: expected parameter declarator
```

jcw, 2020-05-29:
This gets me past the first error, maybe it's not the optimiser but changed std header trickery?

```
#undef memcpy
void *memcpy(void *dst, const void *src, size_t n) {
```

jcw, 2020-05-29:
No, that just moves the problem - 6 #undef's and now I get linker errors:

```
Undefined symbols for architecture i386:
  "___bzero", referenced from:
      _gc_init in gc.o
      _gc_alloc in gc.o
      _gc_realloc in gc.o
      _int_to_bytes in objint.o
      _mp_obj_list_append in objlist.o
      _mp_obj_list_init in objlist.o
      _str_rsplit in objstr.o
      ...
  "___memcpy_chk", referenced from:
      _closure_call in objclosure.o
[...]
```

I don't know what the "_chk" versions are.

dpgeorge, 2020-05-29:
It's still using fortify-source, that's what the "chk" versions are.  You can try running the following to see what config options are defined by default for the compiler:
```
$ gcc -m32 -dM -E - </dev/null
```
(probably a good idea to also add all the other CFLAGS options there to get accurate output)

But otherwise, I see you're trying to build the non-cross minimal port, ie the hosted version.  That probably doesn't need string0.c so you could just try deleting that line from the Makefile.

jcw, 2020-05-29:
Aha, several good tips to go on. I'll report here, just to have this macos/minima/hosted on record.

Without strings0.c

```
LINK build/firmware.elf
ld: warning: The i386 architecture is deprecated for macOS (remove from the Xcode build setting: ARCHS)
ld: warning: ignoring file /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/lib/libSystem.tbd, missing required architecture i386 in file /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/lib/libSystem.tbd
Undefined symbols for architecture i386:
  "___bzero", referenced from:
```

I also had to change to `LDFLAGS = -m32 #-Wl,-Map=$@.map,--cref -Wl,--gc-sections`, as those linker options were not accepted. Maybe 32-bit has become (been made) impossible on MacOS.

No joy so far in finding a way out. I've attached the output of this cmd below:

* $ gcc -m32 -I. -I../.. -Ibuild -Wall -Werror -Wdouble-promotion -Wfloat-conversion -std=c99  -Os -DNDEBUG -fdata-sections -ffunction-sections -DMICROPY_ROM_TEXT_COMPRESSION=1 -ffreestanding -fno-builtin -fno-lto -c -MD -dM -E - </dev/null > defines.txt

My reason for trying the hosted minimal build, is to get a very small set of functions for exploring some VM "trampoline" ideas. Maybe I should just focus on working with the µPy we have today.

I did notice that the hosted minimal version also has trouble when building without the compiler.

[defines.txt](https://github.com/micropython/micropython/files/4701804/defines.txt)

dpgeorge, 2020-05-29:
I can't see anything in the defines.txt about fortify source...

I'm able to build the minimal port as-is (ie x86 32-bit) with clang 10.0.0 on Arch Linux.  So I guess this is a combination of Mac and -m32.  I'd suggest to remove the -m32 (it's not needed, it'll work just fine build as a 64-bit executable).  Or build in a virtual box under a Linux distro (be sure to install the relevant gcc header files).

jcw, 2020-05-29:
Yes, I tried that:

```
CC ../../py/mpprint.c
../../py/mpprint.c:213:13: error: cast to 'mp_obj_t' (aka 'void *') from smaller
      integer type 'unsigned int' [-Werror,-Wint-to-void-pointer-cast]
        x = MP_OBJ_NEW_SMALL_INT(mp_obj_get_int(x));
            ^
../../py/obj.h:90:42: note: expanded from macro 'MP_OBJ_NEW_SMALL_INT'
#define MP_OBJ_NEW_SMALL_INT(small_int) ((mp_obj_t)((((mp_uint_t)(small_int)...
```

Haven't tried to address this yet, because I don't know how deep this rabbit hole might be ("make -k" generates quite a few more error messages).

dpgeorge, 2020-05-29:
Try this:
```diff
--- a/ports/minimal/mpconfigport.h
+++ b/ports/minimal/mpconfigport.h
@@ -69,8 +69,8 @@
 // to print such value. So, we avoid int32_t and use int directly.
 #define UINT_FMT "%u"
 #define INT_FMT "%d"
-typedef int mp_int_t; // must be pointer size
-typedef unsigned mp_uint_t; // must be pointer size
+typedef intptr_t mp_int_t; // must be pointer size
+typedef uintptr_t mp_uint_t; // must be pointer size
 
 typedef long mp_off_t;
```

jcw, 2020-05-29:
Bingo, Damien!

```
$ make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
LINK build/firmware.elf
__TEXT	__DATA	__OBJC	others	dec	hex
114688	4096	0	4295032832	4295151616	10002d000
```

The other changes are:

```
CFLAGS = $(INC) -Wall -Werror -Wdouble-promotion -Wfloat-conversion -std=c99 $(COPT)
LDFLAGS = #-m32 -Wl,-Map=$@.map,--cref -Wl,--gc-sections
```

And I had to remove string0.c from the list, since it keeps fighting with the macros.

For completeness, one more hack now allows building without compiler, in ../../lib/utils/pyexec.c:

    mp_raise_msg(&mp_type_RuntimeError, (void*) "script compilation not supported");

Not tested, merely compiled so far:

```
LINK build/firmware.elf
__TEXT	__DATA	__OBJC	others	dec	hex
81920	4096	0	4295024640	4295110656	100023000
```

Many thanks.

jcw, 2020-05-29:
Off topic, but one final note about the above (not worth an issue, there are already far too many):
with disabled GC, `minimal/main.c` needs an `#if` around gc_collect to compile.

dpgeorge, 2020-05-29:
All these issues should be fixed in #6093