micropython/micropython #6057: meet error when compiling ports/esp32

ceerqingtingml, 2020-05-20:
sdk version:`git describe --tag
v1.12-464-gcae77daf0
`
when I compile esp32 following the README.md in the below link:
[https://github.com/micropython/micropython/tree/master/ports/esp32](ports/esp32)
I met an error:
`CC network_ppp.c
network_ppp.c: In function 'ppp_ifconfig':
network_ppp.c:234:17: error: incompatible types when assigning to type 'const ip_addr_t * {aka const struct ip_addr *}' from type 'ip_addr_t {aka struct ip_addr}'
             dns = dns_getserver(0);
                 ^
../../py/mkrules.mk:63: recipe for target 'build-GENERIC/network_ppp.o' failed
make: *** [build-GENERIC/network_ppp.o] Error 1
`
I checked the code, and found it is correct, please help me on this issue, thanks a lot!

tve, 2020-05-20:
You have to check out the exact esp-idf SHA mentioned in the Makefile (9e70825d1e1cbf7988cf36981774300066580ea7).

chrisovergaauw, 2020-12-04:
@tve I'm running into this issue during an upgrade of my MP setup.

support hashes defined in Makefile:
ESPIDF_SUPHASH_V3 := 9e70825d1e1cbf7988cf36981774300066580ea7
ESPIDF_SUPHASH_V4 := 4c81978a3e2220674a432a588292a4c860eef27b

current ESP-IDF hash is: 9e70825d1e1cbf7988cf36981774300066580ea7
(So that's the first one)

I ran ./install.sh and . ./export.sh

make clean && make erase in the esp32 port results in the same error as described above.

Are there other steps to be taken?

dpgeorge, 2021-04-14:
The esp32 now uses cmake.  Please try with the latest version and if you have errors reopen this issue with further details.

closed 2021-04-14 by dpgeorge

krembuk, 2021-09-13:
stumbled upon this error, and the solution was: 
`git submodule update`