micropython/micropython #6059: Factorizing gc_collect implementation

EmilieNumworks, 2020-05-20:
`gc_collect` implementations have some flaws:
- `gc_collect` is implemented per port. However, parts of its different implementations are duplicated across ports.
- Some helpers are available in `lib/utils`: `gchelper_native.c`, `gchelper_generic.c`. The former uses assembly implementations and the latter uses non-standard C (the `asm` keyword is not required by the C99 specification).
- The configuration `MICROPY_GCREGS_SETJMP`  is not supported by all ports.

I suggest to open a PR in order to :
- Define two new macros in `py/mpconfigport.h`:
    - `MICROPY_STACK_DIRECTION` which indicates whether the stack is ascending or descending
    - `MICROPY_STACK_TOP`: most of the time, it would be defined as `(uintptr_t)MP_STATE_THREAD(stack_top)` but not for ports `esp8266` and `nrf` for instance.
- Implement a generic `gc_collect` in `py/` which relies on the two previous macros and on a function `gc_helper_get_regs_and_sp`. `gc_helper_get_regs_and_sp` will be implemented per port if `!MICROPY_GCREGS_SETJMP` and it will have a default implementation in `py/` if `MICROPY_GCREGS_SETJMP`.
- Implement per port a `gc_helper_get_regs_and_sp` using assembly code to avoid non-standard C. The disjonction which chooses the implementation of `gc_helper_get_regs_and_sp` to use for each port will therefore be in some Makefiles instead of being half in Makefiles and half with C macros in `gchelper_generic.c`.

Let me know if such a PR would be interesting.

dpgeorge, 2020-05-20:
The intention of #5961 was to start a clean-up of the GC collection by factoring some duplicated code.  Further improvements in that direction would definitely be welcome.

To make it easier to review and faster to merge, please try to make PRs as minimal and self contained as possible.  I know it's sometimes not possible to split things up, but for example it might work to have a PR just to introduce `MICROPY_STACK_DIRECTION` and `MICROPY_STACK_TOP`.

> Implement per port a `gc_helper_get_regs_and_sp` using assembly code to avoid non-standard C.

I'm not sure I understand the intention of this part... surely such code could be made available to all ports that need the same implementation to reduce duplication?  There is already `lib/utils/gchelper_m0.s` and `lib/utils/gchelper_m3.s` for Thumb-based archs.

EmilieNumworks, 2020-05-20:
> I'm not sure I understand the intention of this part... surely such code could be made available to all ports that need the same implementation to reduce duplication? There is already lib/utils/gchelper_m0.s and lib/utils/gchelper_m3.s for Thumb-based archs.

Yes, of course. My point was to implement them in assembly rather then in C to avoid non-standard C. It has already be done for `lib/utils/gchelper_m*.s` but we could also split `lib/utils/gchelper_generic.c` in  `lib/utils/gchelper_x8664.s`, `lib/utils/gchelper_i386.s` and so on. Doing so, we'll have to replace the C macros which made the disjunction per architecture by smart Makefiles.

dpgeorge, 2020-05-20:
> My point was to implement them in assembly language rather than in C to avoid non-standard C

The files `py/nlr_*.c` already contain non-standard C in this respect, so I don't see the advantage of doing this.

> Doing so, we'll have to replace the C macros which made the disjunction per architecture by smart Makefiles.

Yes and I think that's the difficult part... at the moment the unix port will compile and work well (with `gc_collect`) on many different archs without any extra effort/config.