micropython/micropython #6063: Problems while building

aam83, 2020-05-21:
Hi!
It's my first time with both ESP32 and Micropython. We are trying to compile it because we will need to add modules in the future, but by the moment it was impossible for us.
We are working in Win10 64 bits.
We installed from Scracth MinGW 32-bits (msys2-i686-20190524.exe download) and added the modules required by windows_install_prerequisites.sh from IDF 4.
We installed ESP IDF 4.0 and re-executed windows_install_prerequisites.sh.
We also installed with PIP the remaining components listed in $IDF_PATH/requirements.txt.
We also executed install.sh and source export.sh in folder $IDF_PATH.
We cloned micropython, and make mpy-cross with some problems: make mpy-cross STRIP=echo SIZE=echo
We created a custom makefile_usr where we added the line "ESPIDF ?= $(HOME)/esp32/esp-idf"
The problems come after make -f makefile_usr:

adrianaznar@PCELEC13 MINGW32 ~/esp32/micropython/ports/esp32
$ make -f makefile_adri
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Building with ESP IDF v4
GEN build-GENERIC/sdkconfig.h
Traceback (most recent call last):
  File "E:/msys32/home/adrianaznar/esp32/esp-idf/tools/kconfig_new/kconfiglib.py", line 1128, in _open
    return open(filename)
FileNotFoundError: [Errno 2] No such file or directory: '/home/adrianaznar/esp32/esp-idf/components/bootloader/Kconfig.projbuild'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "E:/msys32/home/adrianaznar/esp32/esp-idf/tools/kconfig_new/kconfiglib.py", line 1158, in _enter_file
    self._file = self._open(filename)
  File "E:/msys32/home/adrianaznar/esp32/esp-idf/tools/kconfig_new/kconfiglib.py", line 1141, in _open
    raise IOError(
OSError: Could not open '/home/adrianaznar/esp32/esp-idf/components/bootloader/Kconfig.projbuild' (ENOENT: No such file or directory). Perhaps the $srctree environment variable (which was unset) is set incorrectly. Note that the current value of $srctree is saved when the Kconfig instance is created (for consistency and to cleanly separate instances).

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "E:/msys32/home/adrianaznar/esp32/esp-idf/tools/kconfig_new/confgen.py", line 571, in <module>
    main()
  File "E:/msys32/home/adrianaznar/esp32/esp-idf/tools/kconfig_new/confgen.py", line 230, in main
    config = kconfiglib.Kconfig(args.kconfig)
  File "E:/msys32/home/adrianaznar/esp32/esp-idf/tools/kconfig_new/kconfiglib.py", line 653, in __init__
    self._parse_block(None,           # end_token
  File "E:/msys32/home/adrianaznar/esp32/esp-idf/tools/kconfig_new/kconfiglib.py", line 1583, in _parse_block
    self._enter_file(sourced_file)
  File "E:/msys32/home/adrianaznar/esp32/esp-idf/tools/kconfig_new/kconfiglib.py", line 1167, in _enter_file
    .format(self._filename, self._linenr, e.message))
AttributeError: 'OSError' object has no attribute 'message'
make: *** [Makefile:403: build-GENERIC/sdkconfig.h] Error 1
(build-venv)

straga, 2020-05-22:
Try to use WSL on windows, that work.

aam83, 2020-05-22:
Hi! So no one was able to build it under Windows?