micropython/micropython #6065: SSL [Errno -1] ERR_MEM

tyrant666, 2020-05-21:
I'm trying to make a secure connection between esp32 and a website. during the ssl module the memory error is displayed. the generated certificate has the following settings (x509 sha256 rsa: 2048) the error is [Errno -1] ERR_MEM

tve, 2020-05-21:
You're running out of memory... Each TLS connection requires ~20KB of buffers in the esp-idf heap (not the MP heap). If you are not maxing out the MP heap you could make it smaller, leaving more memory to esp-idf. To do so, you have to change the heap allocation done in main.c

dpgeorge, 2020-05-24:
I see this issue was also posted at https://github.com/pycom/pycom-micropython-sigfox/issues/451 .  What board and version of MicroPython where you running when you got the error?

tyrant666, 2020-05-24:
(sysname='LoPy4', nodename='LoPy4', release='1.20.2.rc7', version='v1.11-6d01270 on 2020-05-04', machine='LoPy4 with ESP32', lorawan='1.0.2', sigfox='1.0.1', pybytes='1.4.0')

it is a Pycom board (board L04, 1.20.2.rc7), it is the last release I cannot say the version of the micropyton.

dpgeorge, 2020-05-26:
The firmware you are running is not built from this repository, it is Pycom specific firmware.  So you'll need to ask them for support.

closed 2020-05-26 by dpgeorge