micropython/micropython #6067: RFC: make main.py configurable to support safe mode or versioning

tve, 2020-05-21:
I'm implementing a "safe mode" for my boards that boots a minimal set of functionality so I can recover remote boards when I messed up, e.g. due to uploading a buggy file. The way I'm doing it is by setting `sys.path[:] = ["/safemode", ""]`, i.e., instead of loading modules from the usual `/lib` etc I load from `/safemode/lib` etc. _Except_ for `boot.py` and `main.py`, that is. 

I'm using an ESP32 and there `main.c` basically says "execute /boot.py; execute /main.py", essentially as if it were "concatenate `boot.py` and `main.py` and execute the result". I looked at stm32 and there it's different, `boot.py` gets to set up stuff earlier in the boot process (I'm not really familiar with what that's for). Also, I saw that the `pyb` module has a `pyb.main()` function which can be used in `boot.py` to determine what "main.py" is really called. That would be handy for me because `boot.py` could determine whether to enter safe mode and then `main.py` could come from `/safemode/main.py`...

Questions:
- I have not seen the equivalent of `pyb.main()` in `machine` or `micropython`, is that feature seen as deprecated or not yet ported?
- The substantially same effect could be realized by setting `sys.path` if `main.py` was looked for in the path (and respecting `cur_dir` since that has to be in `sys.path`), has that been considered, would that be an alternative to `pyb.main()`?
- Is all this nonsense and I should determine whether to use safe mode in `main.py` and then load `real_main.py` or `/safemode/real_main.py`? (Except that `main.py` executes in the `__main__` namespace while `real_main.py` wouldn't.)

dpgeorge, 2020-05-22:
> I have not seen the equivalent of `pyb.main()` in `machine` or `micropython`, is that feature seen as deprecated or not yet ported?

It's custom to the `pyb` module, and was never requested on other ports.  I guess it could be made generally available but using `os.chdir()` is probably better.

>  The substantially same effect could be realized by setting `sys.path` if `main.py` was looked for in the path (and respecting `cur_dir` since that has to be in `sys.path`), has that been considered, would that be an alternative to `pyb.main()`?

`main.py` (nor `boot.py`) is not searched for in sys.path because it's an execfile, not a proper import (so it executes in the REPL namespace).  But execfile still respects the current directory so it's possible to do `os.chdir('safemode')` in `boot.py` and then it'll execute `safemode/main.py`.

> Is all this nonsense and I should determine whether to use safe mode in `main.py` and then load `real_main.py` or `/safemode/real_main.py`?

Yes that will also work.  I'd prefer documenting such schemes (chdir or use main.py to launch the true main) instead of adding `pyb.main()` to other ports.

tve, 2020-05-22:
I didn't realize that os.chdir would work for this, thanks for the tip!

tve, 2020-05-23:
Well, I implemented my safe-mode using chdir and I don't think I like it :-(

The reason I don't like it is that it affects all filesystem operations. Specifically, if you have data files say in `/data/foo.db` you now need to use absolute paths to retrieve them. Or if you update the bad code that landed you in safe-mode you need to `pyboard -f cp myapp.py :/` (mind that trailing slash!) or `pyboard -f cp myapp.py :/app/` (mind that leading slash!).

One workaround would be to do an `os.chdir('/')` in main.py to go back, but that's too late for pyboard.py and results in a system where cwd is '/' _except_ when using pyboard.py.

IMHO the loading of main.py should be changed to act the way import does in terms of searching. Another case is that it doesn't support main.mpy as someone recently discovered https://forum.micropython.org/viewtopic.php?f=6&t=8410

jcw, 2020-05-24:
If safemode is for last-resort recovery, then does this really have such an impact? I'd assume that it contains the _absolute minimum_ to reach the unit, and fix things remotely. A bit like running off a recovery partition (which usually also changes the root file system). IOW, isolation as a benefit?

tve, 2020-05-24:
> If safemode is for last-resort recovery, then does this really have such an impact?

It's not the end of the world. But if you start putting files I find it very easy to forget that cwd changed. So far I've never used chdir or seen a need for it, so I'm not used to think about it at all when dealing with the board filesystem.