micropython/micropython #6069: os.chdir to non-existent path

jcw, 2020-05-23:
When doing `os.chdir('/blah')`, an ENODEV error is thrown, as expected.

But a subsequent `os.getcwd()` also throws this error, i.e. the directory was changed anyway.
I expected the working directory to remain as before.

My reason for trying this, is that `os.chdir('/sd'); import app` in `main.py` could be a convenient way to run application code when an SD card is present, while falling back to another file when not.

peterhinch, 2020-05-23:
The Unix build works as expected, but I can confirm this on a Pyboard 1.0.

Re your problem, on a Pyboard **sys.path** changes depending on whether an SD card is present. If one is present it reads
`['', '/sd', '/sd/lib', '/flash', '/flash/lib']`
whereas in the absence it reads
`['', '/flash', '/flash/lib']`.
This means that if present an SD card is searched prior to flash. There are several options for using this behaviour to achieve your aim.

robert-hh, 2020-05-23:
Which board are you using, which filesystem and which firmware version?

jcw, 2020-05-23:
Aha, excellent, the path adjustment is indeed sufficient. Many thanks for pointing this out.
I'll leave this issue open for the "stuck in bad working dir" behaviour on PyBoard etc.

(I'm using `v1.12-464-gcae77daf0-dirty on 2020-05-23; MCUDEV STM32F407ZG` from @mcauser)

dpgeorge, 2020-05-28:
Thanks for the report.  See #6085 for a fix.

closed 2020-05-28 by jcw