micropython/micropython #6070: __slots__ gone?

Pecius, 2020-05-23:
I remember how it used to be implemented but now it appears to be no longer existing in the code.
I tried to search for a reason why it's been removed but I couldn't find anything about it.
In fact, the history of _py/objtype.c_ file doesn't contain the commit it came in.
This is the missing commit 551660df0994b69b915b5b37f27801eb9e0e9ff7 which comes from #3392.

What happened to it?

Pecius, 2020-05-26:
I think I got myself a bit confused here and I must have remembered it wrongly.
Because after further investigation I don't think this feature was even merged into the master branch in the end.

I'm sorry for the confusion caused.

closed 2020-05-26 by Pecius

dpgeorge, 2020-05-27:
That's right, this feature was never merged.

If it's interesting it can be investigated, but it would be low priority.

hlovatt, 2021-07-08:
I would be interested in `__slots__` functionality since it would allow:
```python
class Point(namedtuple('Point', 'x y')):
    __slots__ = ()
    @property
    def d_squared(self):
        return self.x**2 + self.y**2
```
And it would prevent:
```python
p = Point(3, 5)
p.x = 2  # Error!
```
Currently you can enter the above code but it doesn't generate the error on assignment (it makes a new `x` that hides the old one and leads to unexpected behaviour):
```python
>>> p
Point(x=3, y=5)
>>> p.d_squared
34
>>> p.x=4
>>> p
Point(x=3, y=5)
>>> p.d_squared
41
```
Note how the print still says `x = 3` but `d_squared` has changed!

This leads to very hard to find bugs! It prints as though `x` is 3 but behaves with `x = 4`!

This ability to make a field immutable is valuable in safety critical applications, in addition to the RAM saving also noted.

whaleygeek, 2023-02-02:
@dpgeorge I'm doing a big thing on using MicroPython in memory constrained safety sensitive applications (MicroPython in space). 

The extra speed and memory savings provided by ```__slots__``` would be really very useful, as it gets the performance and memory overhead closer to a C build, which is critical on a tiny platform. There is anecdotal evidence that on host based Python, you can get a 20% decrease in execution time, and good savings on memory use. We also use controlled/manual garbage collection for timing predictability, so smaller objects make that much better too.

The extra protection of classes having dynamic extension disabled (for safety reasons) would also be useful. I know we can use a ```__setattr__``` trick for preventing dynamic changes to class membership, but when something is a few million miles away, you don't want any surprises that you haven't experienced in the ground-based testing!

prusnak, 2023-02-02:
@whaleygeek my commit from 2 years ago is still here in case you want to revive it: https://github.com/trezor/micropython/commit/f0d699eb275218daa8bc76e6ec86fc26fa2ea1a6

dpgeorge, 2023-02-16:
@whaleygeek We did do a lot of work on determinism, optimisation and robustness as part of the ESA MicroPython projects.  And most of that work is part of the repo here.  For example, all of the things you can do with the heap locked.

Also the recent addition of caching map lookups has speed up the case of classes with members, so I'm not sure you'd get much of a performance improvement using `__slots__`, although would be interesting to measure it and see if it's worth adding this feature.

whaleygeek, 2023-02-16:
Thanks @dpgeorge I will indeed give it a try. Currently making good progress with viper and other things. I could do with a way to set a maxim time limit on the garbage collector, if you know of any existing work in that area.

dpgeorge, 2023-02-16:
> I could do with a way to set a maxim time limit on the garbage collector, if you know of any existing work in that area.

Do you use `gc.threshold()`?  Try `gc.threshold(4000)` at the start of your code.  It should run the GC often enough to limit the time it runs.

You can also try increasing `MICROPY_ALLOC_GC_STACK_SIZE` (requires a rebuild of the firmware).  If that number is too small then the GC can take twice as long (or longer).  So make it large enough for your application.