micropython/micropython #6071: ESP32 - EN pin when in Deep Sleep State

hpereira98, 2020-05-24:
When my board is in deep sleep state ( machine.deepsleep() ), when I press the EN button the machine.reset_cause() returns the machine.PWRON_RESET instead of the machine.DEEPSLEEP_RESET. Is this supposed to happen? How can I get the machine.DEEPSLEEP_RESET by pressing a button?

I tried the embedded EN button and connecting a physical button to the EN pin.

tve, 2020-05-24:
See https://github.com/micropython/micropython/pull/6003

hpereira98, 2020-05-24:
So, this is not possible yet right?

tve, 2020-05-24:
So there are two answers to your question. The first is that the current code in mainline is fundamentally broken because it calls the wrong function in esp-idf.

The second is that I'm not sure what "the EN button" is. Given what you write I suspect that it's a button connected to the esp32-wroom module's "CHIP_PU/EN" pin, which is the esp32's CHIP_PU pin. That pin controls power to the internals of the esp32, including the RTC domain. Pulling this pin low is indistinguishable from removing power. When the esp32 comes up it's the same as a power-on, which differs from deep-sleep in that deep-sleep (unless special measures were taken) preserves the RTC and RTC memory.

Dunno whether this helps...

dpgeorge, 2020-05-28:
IIRC there is a bug in the esp32 hardware that means it's not possible to get the correct reset cause in some situations.  See eg https://github.com/espressif/esp-idf/issues/494