micropython/micropython #6072: [ py/core/gc] variable alloc from gc not link to collect root

Neutree, 2020-05-25:
In `objfun.c` -> `fun_bc_call`, variable `code_state` alloc from GC heap or stack(define `MICROPY_NO_ALLOCA` to `0` in mpconfigport.h),

https://github.com/micropython/micropython/blob/cae77daf003212684a84b1b3a331d45564a0c286/py/objfun.c#L263

when alloc from stack,it's ok, 
but alloc from GC, it seems not linked to GC collect root pointer section?  `gc_collect` will collect this variable in `mp_execute_bytecode(code_state, MP_OBJ_NULL)` which lead to some fatal error

In my port, if I define `MICROPY_NO_ALLOCA` to 1, the `code_state` is collected by `gc_collect()`, and I tried ESP32 port, define `MICROPY_NO_ALLOCA` to 1,  It looks like running normally, but when I open gc log and print `code_state`  address, we can see the `code_state` was collected by gc even though it's not crash.

This is my test code:
* set GC heap to 32K
* define MICROPY_NO_ALLOCA to 1 in mpconfigport.h

```
import _thread
import time
import gc

def thread1():
    print("enter thread1")
    thread1Cnt = 0
    while True:
        thread1Cnt += 1
        print(gc.mem_free())
        print("====22222====-thread1 cnt", thread1Cnt, ",", thread1)
        #img = [0]*10
        time.sleep_ms(1)

if __name__ == '__main__':
    _thread.start_new_thread(thread1, ())
    threadMainCnt = 0
    while True:
        threadMainCnt += 1
        print("====11111===thread main cnt", threadMainCnt)
        img = [0]*10
        time.sleep_ms(1)
```

Neutree, 2020-05-25:
ohhh code_state belongs to thread stack, so marked in mp_thread_gc_others, so maybe my port exist some error

Neutree, 2020-05-25:
It a bug in `mpthreadport.c`

In ESP32 port, this line:
https://github.com/micropython/micropython/blob/cae77daf003212684a84b1b3a331d45564a0c286/ports/esp32/mpthreadport.c#L86

should be 
```
gc_collect_root(th->stack, (th->stack_len*sizeof(StackType_t)+1024)/sizeof(void*) );
```

In ESP32, `sizeof(StackType_t) == 1` (which in other arch may == 8 or 4), so current code, we use the default stack size(MP_THREAD_DEFAULT_STACK_SIZE) 5K, now actually collect 4K*4=16KBytes stack area, but actually the allocated thread stack is 5K, and the right value we wish shoud be 4K

for other `sizeof(StackType_t) == 4`  or `sizeof(StackType_t) == 8` , it's collect length will less than 4K, so the value in thread stack can not be marked, which lead to the bug above。

Further more, it's better to only collect thread's used stack area but not all of the stack, maybe we can get thread stack top from FreeRTOS API, then collect from stack top to end of stack

dpgeorge, 2020-05-28:
Thanks for the report.  It does look like there's an issue with the thread stack tracing, because the code assumes `sizeof(StackType_t) == sizeof(void*)`, so it's tracing way more memory than needed.  That needs to be fixed (there won't be a bug in the esp32 but in other ports that copy this threading code).

> Further more, it's better to only collect thread's used stack area but not all of the stack, maybe we can get thread stack top from FreeRTOS API, then collect from stack top to end of stack

That might not be possible because the other thread could still be executing and changing its stack pointer.

dpgeorge, 2020-06-02:
See #6104 for a proposed fix.

dpgeorge, 2020-06-05:
Fixed by 621f40b12c272788533dcafcef62ed2a3449bb3b

closed 2020-06-05 by dpgeorge