micropython/micropython #6073: POWER_3 is not the best RF_SETUP record for point-to-point link

potier97, 2020-05-26:
Hello, I'm testing the NRF24L01 with the built drivers, and in the configurations it appears that the best configuration option for the RF is power_3, and I think this is not the case, the best for this is power_0, since it represents the - 18dBm and this is not configured for the Low Noise Amplifier ?, although I know that there are two configurations one PA (Antenna Power) and the other is LNA (Low Noise Amplifier) n and the latter must be the one configured in nrf24l01.py , I changed this configuration as I am telling you and my communication improved remarkably.

  Thank you

peterhinch, 2020-05-26:
The chip datasheet states that bits 2:1 of the RF Setup register determine the transmitter power. The driver sets these bits to 1 which corresponds to 0dBm (1mW). POWER_0 would set these bits to 0, which (as you confirm) corresponds to -18dBm. This corresponds to a transmit power of approximately 1/64mW.

dpgeorge, 2020-05-28:
POWER_3 gives the most power output in the transmitter.  Are you saying that it's better to use POWER_0, even though it's much less power output?

potier97, 2020-05-30:
Yes, I am doing tests with a basic 6 node wsn, using this driver and at the beginning as it is in nrf24l01.py, and the nodes do not reach a meter to send information, I had to join the nodes, and at that point if I send the information, check the documentation a bit and changed this register to POWER_0, then the data was sent over a distance and communication improved, check what  @peterhinch  told me, and a 0dBm register would be the best power for the antenna, in my case was not like that, with -18dBm the antennas reached the best communication.

tve, 2020-05-30:
It's very common that radio nodes placed close together have poor connectivity. This is due to the fact that the receiver gets overloaded and so you have to dial down the power level to the minimum and sometimes even that doesn't do it. If you do your tests with nodes placed at least 4-5 meters apart you start to get a much more realistic picture, but also check the RSSI (I don't know whether the LNA can be dialed up/down on the nrf2401).

peterhinch, 2020-05-30:
@tve While what you say is true in general, I haven't observed this with the nrf24l01. However I've only used it in two node point-to-point links.

In my view there is no issue with the driver.

Most users are likely to want maximum range: the default settings provide the longest range for a point-to-point link. Users with other needs can use the `set_power_speed` method to change these settings.

rolandvs, 2020-06-01:
It can be done, worked on Myrianed, we were able to show a full self-organizing network of 1024 nodes that were organized in a cube with a node to node distance of 50cm.

peterhinch, 2020-06-02:
Yes. My problem (and I suspect that of @potier97) was as follows. Node A waits for a packet from node B. When this is received node A stops listening and transmits a response. Node B (which has just sent the packet) starts listening for the response. This works fairly well at moderate range, worse at short range. The reason is that when node A sends its response, sometimes node B's receiver is not yet effectively listening.

The solution was for both nodes to pause before transmission to allow the peer's receiver time to initialise. I used 10ms which completely eliminated timeouts (tested at >= 30cm).

This range-dependent delay is probably a chip problem. Receiver AGC settling time perhaps? I'll re-read the datasheet.

In conclusion my opinion is that the driver defaults are correct.

peterhinch, 2020-06-07:
For the benefit of anyone using this chip and driver I have documented some of the issues I encountered [here](https://github.com/peterhinch/micropython-radio/blob/master/async/README.md#12-notes-for-protocol-designers).

In summary:
 1. I found no fault with the driver. There are issues to consider when using it.
 2. There may be an issue with the chip. Under conditions of extremely poor reception it can return corrupt data despite the 16-bit CRC. Occurrences are exceptionally infrequent. I've finally found the reason why my previous attempt to write a rock-solid protocol for this chip suffered rare message loss, something that has bugged me for several years.