micropython/micropython #6078: Is there a port for ARM Cortex-M33(dual-core) board?

xidameng, 2020-05-26:
I wanna work on porting micropython to Realtek Ameba D (RTL8722DM) dev. board but wonder if there is already a similar port I can reference to. If not, where should I start for porting to a new board?

glennrub, 2020-05-26:
Hi @xidameng ,

The `pca10090` board (nrf9160) in the `nrf-port` is a cortex-m33 target. I'm not sure how similar this is to your target, but at least something you can take a look at.

xidameng, 2020-05-27:
> Hi @xidameng ,
> 
> The `pca10090` board (nrf9160) in the `nrf-port` is a cortex-m33 target. I'm not sure how similar this is to your target, but at least something you can take a look at.

Thanks a lot! I will take a look~

closed 2020-05-27 by xidameng