micropython/micropython #6086: mem_info - the verbose mode is no different from the normal one

fizista, 2020-05-28:
```
>>> micropython.mem_info(1)
stack: 2112 out of 8192
GC: total: 37952, used: 12112, free: 25840
 No. of 1-blocks: 286, 2-blocks: 22, max blk sz: 264, max free sz: 1609
GC memory layout; from 3ffeedc0:
00000: MDThhhDMhTDhThDTB=TB=BhhTTB=h====h===hh==h======================
00400: ================================================================
00800: ================================================================
00c00: ================================================================
01000: =================================================DhTh=hTTMDThh==
01400: =====h=====TTDhTMhMDTDLTBTBBBBMDhh===MDMTDhhh=B=MDBhBBhBh==h==MD
01800: h=BBBBh=====h=======SB=hTh===B=MDBhBTDhBh========DBBTBBh==h===BB
01c00: BB=hBTTDDDTLh=====hhDTBh=h===BTB=DhThTTDTBh=h===BBh===========h=
02000: ====DBhTBh==h===h==TDhThTBBThhh=BTDhhThTh=Thh==Thh==TDhThTTThTTD
02400: hThBTTTh=hDh========hh=hhhhThTDDh=hhhhhhThTh=h=TThhThTDhThTTThh=
02800: ==========h=======================TTDhThTTThTTThTTDhThDhh=B..h..
02c00: ..h=TTThTTDhThTTThTTDhThTTThTTDhThTTThTTDhThTTThTTDhThTTThT.....
       (25 lines all free)
09400: ....

>>> micropython.mem_info(0)
stack: 2112 out of 8192
GC: total: 37952, used: 8784, free: 29168
 No. of 1-blocks: 118, 2-blocks: 21, max blk sz: 264, max free sz: 1674
GC memory layout; from 3ffeedc0:
00000: MDThhhDMhTDhThDTB=TB=BhhTTB=h====h===hh==h======================
00400: ================================================================
00800: ================================================================
00c00: ================================================================
01000: =================================================DhTh=hTTMDThh==
01400: =====h=====Th=hBM.MD.DL.B.BBBBMD.h===MDM.Dhhh=B=MDB.BB.Bh==h==MD
01800: ..BBBB..........h=...B=..h===B=MDB.B...Bh========DBB.BBh==h===BB
01c00: BB=.B..DDD.Lh=====h.D.Bh=h===B.B=......D.Bh=h===BBh===========h=
02000: ====DB..Bh==h===h==......BB.hhh=............h==.hh==............
02400: ...B...h=hDh========hh=hhh..h..Dh=h.hhhh...h=h=..h..h.........h=
02800: ==========............................................Dhh=......
       (26 lines all free)
09400: ....

```

tested on: esp32 esp8266 micropython v1.12

dpgeorge, 2020-05-28:
For non-verbose mode don't pass any args: `micropython.mem_info()`

fizista, 2020-05-28:
Thank you. It wasn't that obvious to me.

closed 2020-05-28 by fizista