micropython/micropython #6087: MICROPY_EVENT_POLL_HOOK  and MICROPY_STACKLESS

jcw, 2020-05-28:
The current implementation relies on MICROPY_EVENT_POLL_HOOK in several places. This is akin to event-based applications where you could "call" the event loop to "get more work done". It nests on the C stack. It also prevents MICROPY_STACKLESS from ever becoming implementable, I think.

As I understand it, this has major implications for asyncio. I suspect that a stackless approach would allow async/await to be used even in synchronous Python code, and solve the [red & blue problem](https://forum.micropython.org/viewtopic.php?f=2&t=8429&p=47919#p47850).

My question is: what is the status of MICROPY_STACKLESS? Is there a roadmap for µPy?

PS. This is a tough problem, but not unsolvable, if my understanding is correct. Both Perl and Tcl have a "trampoline" mechanism in their VM main loops to implement this functionality.

peterhinch, 2020-05-28:
Bear in mind that a key goal for **uasyncio** is compatibility with CPython's **asyncio**.

jcw, 2020-05-28:
Yep. This can be 100% compatible and be better (e.g. I/O & libraries being async under the hood).

dpgeorge, 2020-05-28:
I'm not sure exactly what the issue is here, I guess this is more of a discussion around stackless+uasyncio.

> what is the status of MICROPY_STACKLESS?

It works.  When enabled it passes all tests.  In combination with `MICROPY_ENABLE_PYSTACK` it's actually quite efficient because it doesn't need to use the heap to allocate the call frames.

jcw, 2020-05-28:
Thanks. Yes, I admit, it's probably more a discussion topic than an _issue_. It felt like this place was better suited to bring up a topic about implementation.

I'm puzzled. If MICROPY_STACKLESS works, why isn't it the default? Platform-specific issues?

As far as I'm concerned, µPy already works splendidly, but I've voiced my concerns on the forum about async/await creep. I'd be interested to help discuss and even try/code/test changes which (IMO) might help untangle VM / task state, scheduling, polling, and IRQ / event handling. Maybe it's a pipe dream, but if the VM drops back out to the core scheduler whenever anything needs to be done which could block (and that includes all selects & polling), then it should be possible to suspend a task in any "await" call site, even inside synchronous code. IOW, the VM is a function that can be called by the outer scheduler to execute a run of instructions.

Maybe I've got it all wrong. Or maybe MICROPY_STACKLESS already works this way?

Please accept that I'm not trying to bring up academic issues or waste anyone's time. If there's no agreement, or no interest in this for now, then I suggest closing this issue.

dpgeorge, 2020-05-29:
> If MICROPY_STACKLESS works, why isn't it the default? Platform-specific issues?

It's very slow without `MICROPY_ENABLE_PYSTACK` enabled because all call frames are allocated on the heap, and such allocation is expensive.  With `PYSTACK` there's a special Python stack which is independent to the C stack from which these call frames can be linearly allocated and freed in a LIFO manner.

> Maybe it's a pipe dream, but if the VM drops back out to the core scheduler whenever anything needs to be done which could block

The uasyncio core scheduler is actually implemented in Python, and it calls C functions to do the blocking, so there are nested C functions and Python functions going on at the point of blocking.

> then it should be possible to suspend a task in any "await" call site, even inside synchronous code

It would be technically possible to do this but that would require an RTOS/OS and threading, it's just that it would be cooperative task switching rather than preemptive.

> Or maybe MICROPY_STACKLESS already works this way?

For any Python function that calls another Python function, in stackless mode the VM will allocate a new call frame for the called function and just continue executing bytecode.  Upon return it frees the call frame and switches back to the previous call frame.  This eliminates VM->VM calls at the C level and so doesn't eat up C stack.

jcw, 2020-05-29:
Ok. Note that MICROPY_STACKLESS + MICROPY_ENABLE_PYSTACK is not really stackless: the Python VM stack state has moved to another stack, but it can't be re-used after a Task switch. Then again, MICROPY_STACKLESS by itself really is, with the overhead of frames-on-the-heap. Very nice.

If the VM were to return to an upper level ("trampoline") instead of calling the blocking C code itself, the C / Py stack entanglement could be avoided, at some future stage perhaps.

But after a good night's sleep and some further reading, I realise that my current understanding of all this is too weak. I'll close the issue before conflating these issues further. For my own use, I've coded a "minimally async" MQTT client and web/websocket server, which is good enough for now.

Thank you for clarifying.

closed 2020-05-29 by jcw

jcw, 2020-05-31:
For reference, I'd like to follow up on where the C and Py stacks mix, ***IF*** one day someone does decide to reorganise it.

In `uasyncio.sleep(_ms)`, a task gets pushed on the priority queue [here](https://github.com/micropython/micropython/blob/22806ed5df27c10131af0cedb2f7e8b134fe6e7a/extmod/uasyncio/core.py#L44), leading to wait_io_event [here](https://github.com/micropython/micropython/blob/22806ed5df27c10131af0cedb2f7e8b134fe6e7a/extmod/uasyncio/core.py#L167), all in Python code right now. This in turn calls uselect's ipoll [here](https://github.com/micropython/micropython/blob/22806ed5df27c10131af0cedb2f7e8b134fe6e7a/extmod/uasyncio/core.py#L113), at which point a C stack frame is created. Since ipoll loops on MICROPY_EVENT_POLL_HOOK [here](https://github.com/micropython/micropython/blob/master/extmod/moduselect.c#L250-L257), that's where the entanglement happens, as I understand it.

To disentangle, a _minimal_ scheduling core would have to be written in C, so that the VM can drop out of its interpreter loops, at the point where it now calls uselect's `ipoll`, with an outer "mini scheduling loop" performing this task instead. That would allow a Py task context switch to happen, and then to resume the VM in that new context.

dpgeorge, 2020-06-02:
> that's where the entanglement happens, as I understand it.

Yes that's correct, it's currently in the `ipoll()` function which does the low level "waiting for event".

> To disentangle, a _minimal_ scheduling core would have to be written in C, so that the VM can drop out of its interpreter loops,

Another way might be to have a new "wait for event" opcode in the VM...