micropython/micropython #6088: Unable to catch MemoryError in ISR

andrewleech, 2020-05-28:
I'm not sure if this is expected behavior, or a bug?
``` python
>>> def locked():
...     try:
...         _ = bytearray(1)
...         print(False)
...         return
...     except MemoryError:
...         pass
...     print(True)

>>> locked()
False
>>> import micropython
>>>micropython.heap_lock();locked()
True

>>> from pyb import Timer
>>> tim = Timer(5)
>>> tim.init(freq=1, callback=locked)
>>> uncaught exception in Timer(5) interrupt handler
Traceback (most recent call last):
  File "logger/logger.py", line 33, in heap_locked
MemoryError: 
```

davehylands, 2020-05-28:
If you add the following two lines:
```
import micropython
micropython.alloc_emergency_exception_buf(100)
```
then you'll get the real error:
```
>>> uncaught exception in Timer(5) interrupt handler
TypeError: function takes 0 positional arguments but 1 wer
```
i.e. your code was never being executed. If you add an argument to the locked function (timer callbacks are passed a timer object), then it runs with no exceptions being printed.

davehylands, 2020-05-28:
What's happening is that because the heap is locked, trying to allocate memory for the TypeError exception fails, causing the MemoryError. Essentially, ALL exceptions from within ISR context cause a MemoryError while trying to allocate exception object, unless you have the emergency_exception buffer allocated.

andrewleech, 2020-05-29:
Ah, of course. Unfortunately this was supposed to be a repro of this issue I've seen in my real project, where the locked function is being used from within other, larger isr functions that otherwise run fine.

Interestingly, I do already have micropython.alloc_emergency_exception_buf(120) in the same project, which would have been run before I jumped to repl to trial the code above - so I'm not sure why I wasn't getting the real exception printed. I'll have to do some more testing...

andrewleech, 2020-05-29:
Ok, so the error first reported was more complicated than I first realised, as we've got a logger module based on `uos.dupterm`. That's where the actual traceback above comes from `  File "logger/logger.py", line 33, in heap_locked`

I've got a more complete reproduction case now I believe, but it's a tad messy:
``` python
import uio
import uos
import pyb

def heap_locked():
    try:
        _ = bytearray(1)
        return False
    except MemoryError:
        pass
    return True

class LoggerImpl(uio.IOBase):

    def __init__(self):
        self.stdout = pyb.USB_VCP(0)
        uos.dupterm(self, 0)

    def write(self, buf):
        in_interrupt = heap_locked()
        if in_interrupt:
            self.stdout.write("INT")

Logger = LoggerImpl()

from pyb import Timer
tim = Timer(5)
tim.init(freq=1, callback=lambda: None)
```
When run, causes:
```
>>> INTuncaught exception in Timer(INT5INT) interrupt handlerINT
INTTraceback (most recent call last):INT
INT  File "INTlogger/logger.pyINT", line INT29INT, in INTheap_lockedINT
INTMemoryErrorINT: INT
```
So the same unabled to catch MemoryError - just with bits of INT sprinkled from the dupterm logger.

I've only managed to trigger this problem by having:
* a dupterm handler in place which tries to check/catch MemoryError 
* an exception occurs in an interrupt handler, from which the stack track is printed to stdout which flows through to the dupterm handler.

davehylands, 2020-05-29:
This is basically the same problem as your first example. I added the emergency exception buffer and I get this error:
```python
>>> dupterm: Exception in read() method, deactivating: AttributeError: 'LoggerImpl' object has no attribute 'readinto'
uncaught exception in Timer(5) interrupt handler
TypeError: function takes 0 positional arguments but 1 were given
```
Your `lambda: None` needs to take a single argument. When I change it to be `lambda t: None` then I just get this (on my pyboard:
```python
>>> dupterm: Exception in read() method, deactivating: AttributeError: 'LoggerImpl' object has no attribute 'readinto'
```

andrewleech, 2020-05-29:
Whoops, I did have an stub readinto in the copy I tested, trimmed a little too much for the post - sorry. I've added the readinto function to the example in my post just above

I definitely do have `micropython.alloc_emergency_exception_buf(100)` enabled (just re-ran it to be sure) and I get the same issue.

I should have mentioned, I left that lambda without an arg on purpose to trigger the problem - the issue only shows itself when it's an exception from interrupt handler printing via the dupterm.

davehylands, 2020-05-29:
I guess I don't see the problem. I can reproduce your results. There is an uncaught exception in Timer 5 because the lambda function doesn't have an argument.

The INT calls aren't coming from your handler. The INT prints are coming from trying to print the
```
uncaught exception in Timer(5) interrupt handler
TypeError: function takes 0 positional arguments but 1 were given
```
which is occurring from ISR context (basically the timer ISR has a try/except around the call to tg=he callback. This is the try:
https://github.com/micropython/micropython/blob/1662a0b06f2ad58aa4a632a88d0b3836c08bded2/ports/stm32/timer.c#L1597
and this is the except:
https://github.com/micropython/micropython/blob/1662a0b06f2ad58aa4a632a88d0b3836c08bded2/ports/stm32/timer.c#L1602-L1610

The print happens on the last line of the except. The heap is still locked while the print occurs, so the write function in your logger is called with the heap locked.

If  I add the argument, then no exceptions occur from an ISR context, and no calls to write are made.

davehylands, 2020-05-29:
If anything, the fact that there are INT's being printed demonstrates that the try/catch inside your heap_locked function is actually working properly.

andrewleech, 2020-05-29:
Yes, I agree with what you're saying, I do understand the failing exception comes from the timer lambda. 

That's a good point that the heap_check function is clearly working, in that the INT bits are printed out - the confusion was from the printed exception being 
```
Traceback (most recent call last):
  File "logger/logger.py", line 33, in heap_locked
MemoryError: 
```
but now thinking about it, because this is all running in heap locked state the exception stack trace in `heap_locked()` must be getting rendered into the `alloc_emergency_exception_buf` even though the exception is handled cleanly, overwriting the original timer exception stack trace.

So the only "issue" as such is that the logged exception is a distraction, hiding the original cause of the failure. Probably not much can/should be done about that as I presume the stack trace needs to get written to that buffer long before the runtime knows if the exception is handled or not.

davehylands, 2020-05-29:
OK - I get what you're seeing now. The exception buffer will get overwritten if another exception occurs between the time the original exception happens and the original exception getting printed.

This is because emergency exception buffer is treated like a global, rather than a mini-heap.

andrewleech, 2020-05-29:
Yep that makes perfect sense to me know. I've been seeing this error off and on for many months now and was pulling my hair out a bit wondering why MemoryError was escaping the try/catch.

It's great to know it actually isn't, that there isn't any sneaky bug here at all, just behavior that can be somewhat confusing. I wonder where would be a good place to document this to help others...

Thanks so much for working through this @davehylands I really appreciate the help!

andrewleech, 2020-06-01:
Closing this issue as the confusion has been explained. I've opened a PR (linked above) with a suggested additional paragraph in the docs to try to warn users of this.

closed 2020-06-01 by andrewleech