micropython/micropython #6089: getattr not falling back to default with custom __getattr__

dnr, 2020-05-28:
Here's an example:

```
class tester:
  def __init__(self):
    self.data = {'a': 5}

  def __getattr__(self, n):
    if n in self.data:
      return self.data[n]
    else:
      raise AttributeError(n)

t = tester()

print(getattr(t, 'a', 'default'))
print(getattr(t, 'b', 'default'))
```

In CPython, that prints `5`, `default`. In MicroPython (1.12), it prints `5`, but then the `AttributeError` gets raised to the top level and `getattr` doesn't return `default`.

dpgeorge, 2020-05-29:
Thanks for the report, indeed it is an inconsistency with CPython.

See #6090 for a proposed fix.

dpgeorge, 2020-06-02:
Fixed by 203b10703e5df20939b88a2ce29bb961cce7e4b6

closed 2020-06-02 by dpgeorge