micropython/micropython #6094: metaclass

allyourcode, 2020-05-30:
I see that MP does not (yet) support metaclasses (per item #6 on https://github.com/micropython/micropython/wiki/Differences#differences-by-design). It sounds like the MP developers would like to add this feature some day, but it's not clear to me how strongly they feel about that.

Obviously, people do not directly make use of this feature very often. Nevertheless, it is very powerful, and people DO benefit from libraries that do use it directly. I am working on a library that works in CPython, but because it hinges on a metaclass, I was disappointed to find that it doesn't work on MP.

My hopes in posting this are to

a) Solicit the thoughts of the MP developers on where this feature lies in their priority queue.
b) Let others provide "me too!" feedback.
c) Vent my frustration.

PS: Thank you for MP! Being able to use (u)asyncio is truly a godsend.

dpgeorge, 2020-06-02:
> Obviously, people do not directly make use of this feature very often. Nevertheless, it is very powerful, and people DO benefit from libraries that do use it directly.

Yes, that's also my understanding, that metaclasses are not really a user-facing feature but rather useful to implement things that are user facing.

> Solicit the thoughts of the MP developers on where this feature lies in their priority queue.

For myself, it's very low priority, and we've not really had any requests for it in the past.  The main deciding factor for features like this is not so much how hard it is to add, but rather how much it makes the core interpreter grow in terms of size and complexity.

A good first task would be to write a few small test cases which show how metaclasses should behave.

winni2k, 2024-03-01:
I'm looking at porting dataclasses to MP, and I see that the CPython 3.8 implementation of dataclasses relies on metaclasses (https://github.com/python/cpython/blob/4d58a1d8fb27048c11bcbda3da1bebf78f979335/Lib/dataclasses.py#L206).

bartek-w, 2025-10-28:
I believe this is mostly about easy code portability. It would be great to be able to easily implement a fully functional Enum in MicroPython directly (or similar classes). Maybe MicroPython could support some basic metaclass functionality or add support for __init_subclass__, which would allow implementing substitutions for most basic metaclass functionality?

The metaclasses support is the main blocker when I want to easily copy&paste existing python code and get it running. What are the main blockers to add some metaclass functionality? Added implementation complexity? Memory or CPU overhead? Or metaclasses are simply too much for Micropython from the project design assumptions, as it should be kept 'micro'?