micropython/micropython #6095: qemu-arm fails to build with optimizations

bwasim, 2020-05-30:
*Observation*
My original intent is to use micropython on the STM3L4 series (either bare-metal or zephyr rtos), but because that's a constraint target (1M flash partitioned into 512K code + 512K FS, 96 RAM), I started out with qemu-arm (mps2-an385) to understand more about the internals of micropython. 

I was able to build and run it.. It exits after printing out "hello world" on the virtual console. However, to get more into it, I decided to disable optimizations by modifying the `ports/qemu-arm/makefile` to add this, 
```
# Enable debugging
DEBUG = 1
``` 

By enabling versbose mode, I was able to ensure that this flag does take effect and now the qemu-arm port builds with `O0` optimizations, but linker throws the following error, 
```
bwasim@DESKTOP-G27NF4S:/mnt/c/Users/Bilal/Desktop/wsl/micropython/micropython/ports/qemu-arm$ make -j32
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
GEN build/genhdr/qstr.i.last
GEN build/genhdr/qstr.split
GEN build/genhdr/compressed.split
GEN build/genhdr/qstrdefs.collected.h
GEN build/genhdr/compressed.collected
QSTR not updated
Compressed data not updated
CC main.c
arm-none-eabi-ld: build/py/gc.o: in function `gc_free':
/mnt/c/Users/Bilal/Desktop/wsl/micropython/micropython/ports/qemu-arm/../../py/gc.c:588: undefined reference to `__assert_func'
arm-none-eabi-ld: /mnt/c/Users/Bilal/Desktop/wsl/micropython/micropython/ports/qemu-arm/../../py/gc.c:590: undefined reference to `__assert_func'
arm-none-eabi-ld: build/py/gc.o: in function `gc_realloc':
/mnt/c/Users/Bilal/Desktop/wsl/micropython/micropython/ports/qemu-arm/../../py/gc.c:750: undefined reference to `__assert_func'
arm-none-eabi-ld: /mnt/c/Users/Bilal/Desktop/wsl/micropython/micropython/ports/qemu-arm/../../py/gc.c:686: undefined reference to `__assert_func'
arm-none-eabi-ld: /mnt/c/Users/Bilal/Desktop/wsl/micropython/micropython/ports/qemu-arm/../../py/gc.c:688: undefined reference to `__assert_func'
arm-none-eabi-ld: build/py/mpprint.o:/mnt/c/Users/Bilal/Desktop/wsl/micropython/micropython/ports/qemu-arm/../../py/mpprint.c:208: more undefined references to `__assert_func' follow
make: *** [Makefile:140: build/firmwa```

*Fix*
Looking at the other ports, this problem is arising because of the missing "weak" definitions of `__assert_func / __fatal_error` in the qemu-arm `main.c` file.. Adding these two functions fixes the problem. 

I'm not sure if this port is intended to work without optimizations or not (i would expect it to), and hence not opening a pull request directly. I'm adding a patch which fixes the problem (simple copy-paste from other ports).. 

[0001-qemu-arm-port-fails-with-optimizations-disabled.txt](https://github.com/micropython/micropython/files/4705916/0001-qemu-arm-port-fails-with-optimizations-disabled.txt)

dpgeorge, 2020-06-02:
Thanks for the report.  Yes this port should build and run with `DEBUG=1`.  See #6103 for a fix which calls exit(1) after printing the assertion failure.

bwasim, 2020-06-02:
Thanks.. 

Looking into the qemu-arm port, it seems to simply print out `hello_world` on the console and exit. Instead, I would expect it to look into the FS for `boot.py` / `main.py`, execute them and fire up REPL, i.e. act like a normal micropython platform in addition to being used for CI testing.. This would allow me to fully analyze micropython without requirement of a physical board.. 

Is something like this being planned for `qemu-arm` at this moment ? If not, I would take some time and add this support, if you think it makes sense..

dpgeorge, 2020-06-02:
> Is something like this being planned for `qemu-arm` at this moment

No plans.

> If not, I would take some time and add this support, if you think it makes sense..

It does make sense.  Feel free to open a PR adding such support.

dpgeorge, 2020-06-05:
Original issue here is fixed by 596fb73927df4f9202729ff9d92a1db479a95ed7

closed 2020-06-05 by dpgeorge