micropython/micropython #6097: Reference target errors when building docs

nanjekyejoannah, 2020-05-31:
While working on a documentation patch, I got a few errors not related to my work when building the docs.

How to reproduce :

```
cd docs
make html
```

Here is the error ;

/```
workspace/micropython/docs/library/micropython.rst:102: WARNING: 'any' reference target not found: MICROPY_PY_MICROPYTHON_HEAP_LOCKED
/workspace/micropython/docs/reference/pyboard.py.rst:70: WARNING: 'any' reference target not found: --baudrate
```

nanjekyejoannah, 2020-06-03:
Upgrading sphinx actually exposes more warnings as stated note [here](https://github.com/micropython/micropython/pull/6099):

Generating MicroPython Differences.
rm -f genrst/*
cd ../tools && python3 gen-cpydiff.py
sphinx-build -E -b html -d build//doctrees   . build//html
Running Sphinx v3.1.0+
loading intersphinx inventory from https://docs.python.org/3.5/objects.inv...
building [mo]: targets for 0 po files that are out of date
building [html]: targets for 158 source files that are out of date
updating environment: [new config] 158 added, 0 changed, 0 removed
reading sources... [100%] wipy/tutorial/wlan                                                                                                            
/workspace/micropython/docs/library/network.WLAN.rst:104: WARNING: duplicate object description of network.WLAN.config, other instance in library/networ
k.WLAN, use :noindex: for one of them
/workspace/micropython/docs/library/pyb.Flash.rst:28: WARNING: duplicate object description of pyb.pyb.Flash, other instance in library/pyb.Flash, use :
noindex: for one of them
/workspace/micropython/docs/library/pyb.Flash.rst:38: WARNING: duplicate object description of pyb.Flash.readblocks, other instance in library/pyb.Flash
, use :noindex: for one of them
/workspace/micropython/docs/library/pyb.Flash.rst:40: WARNING: duplicate object description of pyb.Flash.writeblocks, other instance in library/pyb.Flash, use :noindex: for one of them
/workspace/micropython/docs/library/uio.rst:116: WARNING: duplicate object description of uio.StringIO, other instance in library/uio, use :noindex: for one of them
/workspace/micropython/docs/library/uio.rst:117: WARNING: duplicate object description of uio.BytesIO, other instance in library/uio, use :noindex: for one of them
/workspace/micropython/docs/library/uos.rst:255: WARNING: duplicate object description of uos.AbstractBlockDev.readblocks, other instance in library/uos, use :noindex: for one of them
/workspace/micropython/docs/library/uos.rst:270: WARNING: duplicate object description of uos.AbstractBlockDev.writeblocks, other instance in library/uos, use :noindex: for one of them
looking for now-outdated files... none found
pickling environment... done
checking consistency... done
preparing documents... done
writing output... [100%] wipy/tutorial/wlan                                                                                                             
templates/replace.inc:7: WARNING: 'any' reference target not found: CPython
templates/replace.inc:7: WARNING: 'any' reference target not found: CPython
templates/replace.inc:7: WARNING: 'any' reference target not found: CPython
/workspace/micropython/docs/library/index.rst:26: WARNING: 'any' reference target not found: MicroPython port
/workspace/micropython/docs/library/index.rst:36: WARNING: 'any' reference target not found: MicroPython port
templates/replace.inc:7: WARNING: 'any' reference target not found: CPython
/workspace/micropython/docs/library/micropython.rst:102: WARNING: 'any' reference target not found: MICROPY_PY_MICROPYTHON_HEAP_LOCKED
/workspace/micropython/docs/library/network.rst:41: WARNING: 'any' reference target not found: MicroPython port
/workspace/micropython/docs/library/network.WLAN.rst:119: WARNING: 'any' reference target not found: MicroPython port
templates/replace.inc:7: WARNING: 'any' reference target not found: CPython
templates/replace.inc:7: WARNING: 'any' reference target not found: CPython
templates/replace.inc:7: WARNING: 'any' reference target not found: CPython
templates/replace.inc:7: WARNING: 'any' reference target not found: CPython
templates/replace.inc:7: WARNING: 'any' reference target not found: CPython
templates/replace.inc:7: WARNING: 'any' reference target not found: CPython
/workspace/micropython/docs/library/uerrno.rst:9: WARNING: 'any' reference target not found: MicroPython port
/workspace/micropython/docs/library/uerrno.rst:17: WARNING: 'any' reference target not found: MicroPython port
templates/replace.inc:7: WARNING: 'any' reference target not found: CPython
templates/replace.inc:7: WARNING: 'any' reference target not found: CPython
templates/replace.inc:7: WARNING: 'any' reference target not found: CPython
templates/replace.inc:7: WARNING: 'any' reference target not found: CPython
templates/replace.inc:7: WARNING: 'any' reference target not found: CPython
templates/replace.inc:7: WARNING: 'any' reference target not found: CPython
/workspace/micropython/docs/library/ure.rst:141: WARNING: 'any' reference target not found: MicroPython port
/workspace/micropython/docs/library/ure.rst:145: WARNING: 'any' reference target not found: MicroPython port
/workspace/micropython/docs/library/ure.rst:187: WARNING: 'any' reference target not found: MicroPython port
/workspace/micropython/docs/library/ure.rst:196: WARNING: 'any' reference target not found: MicroPython port
/workspace/micropython/docs/library/ure.rst:202: WARNING: 'any' reference target not found: MicroPython port
templates/replace.inc:7: WARNING: 'any' reference target not found: CPython
templates/replace.inc:7: WARNING: 'any' reference target not found: CPython
/workspace/micropython/docs/library/usocket.rst:37: WARNING: 'any' reference target not found: MicroPython port
/workspace/micropython/docs/library/usocket.rst:37: WARNING: 'any' reference target not found: MicroPython Unix port
[... cut ...]
/workspace/micropython/docs/reference/packages.rst:146: WARNING: 'any' reference target not found: MicroPython port
/workspace/micropython/docs/reference/packages.rst:162: WARNING: 'any' reference target not found: MicroPython port
/workspace/micropython/docs/reference/packages.rst:171: WARNING: 'any' reference target not found: MicroPython Unix port
/workspace/micropython/docs/reference/packages.rst:284: WARNING: 'any' reference target not found: MicroPython Unix port
/workspace/micropython/docs/reference/pyboard.py.rst:70: WARNING: 'any' reference target not found: --baudrate

dpgeorge, 2020-06-05:
All warnings should now be fixed by 1e6d18c915ccea0b6a19ffec9710d33dd7e5f866

closed 2020-06-05 by dpgeorge