micropython/micropython #6105: uasyncio V3 stream readline: is this correct?

peterhinch, 2020-06-02:
The [readline method](https://github.com/micropython/micropython/blob/e54626f4c1b5cb4de0b5dc86d820bb36c1d6499f/extmod/uasyncio/stream.py#L33) returns if the final character is a newline. However it is possible that **self.s.readline()** might return more than one line. Should **Stream.readline** find the first newline and return a partial result? [CPython docs](https://docs.python.org/3/library/asyncio-stream.html):

> Read one line, where “line” is a sequence of bytes ending with \n.

In the existing design the onus is on the device **s** to return a maximum of one line. Is this correct?

dpgeorge, 2020-06-02:
>  However it is possible that **self.s.readline()** might return more than one line.

I'm not sure this should happen.  It really should be required by the underlying stream to return at most one line (or less than one line if it detects end-of-stream).

peterhinch, 2020-06-02:
That's how I'm handling it. My doubt was whether this was a requirement or a workround. If it's a requirement I'll document it in my tutorial.

dpgeorge, 2020-06-03:
If it makes sense, I'd prefer for it to be a requirement for the underlying stream, because this would simplify the implementation of `readline()`.

But it could also make more sense for all the `readline()` logic to be handled by uasyncio, so the underlying stream only needs to implement `readinto()`.... well, for built-in streams like sockets/files it's best to let the socket/file do the readline logic because it's in C and is efficient.  But for user streams it might be better for uasyncio to do it.

I think CPython has the latter behaviour, that readline pulls bytes from the underlying stream using read.

peterhinch, 2020-06-03:
>  But for user streams it might be better for uasyncio to do it.

Given that the `uasyncio` code is in Python would there be much performance advantage either way?

The relevant part of my `.readline` amounts to 6LOC. I'm agnostic on this apart from a desire to know the interface for my code and doc. I'll await your decision.

peterhinch, 2020-06-04:
After further thought,

I remain agnostic over whether uasyncio readline splits multiple lines.

I do think it should continue to use **stream.readline** and not **stream.read** as its data source. This is because it allows the user greater flexibility. Consider this example.

In my application (a resilient protocol for nRF24l01) a line can comprise multiple consecutive packets. If **stream.readline** receives a packet not containing `b'\n'` it buffers it and returns `b''`. When it receives a subsequent packet with the newline, the complete line is returned.

This simplifies application code which never needs to deal with partial lines.

In my case this is a detail rather than a show-stopper, but there may be other scenarios I haven't thought of where special behaviour of **stream.readline** really matters. We don't want to tamper with **.read**.