micropython/micropython #6112: ESP32: Question about thread data structure

robert-hh, 2020-06-05:
I'm wondering about the data element 'ready' in the thread administrative data structure. As far as I can tell, it is only used in mp_thread_gc_others() to tell, whether gc_collect_root() it performed on the threads stack. But in the ESP32 port the stack buffer is provided by RTOS and is itself not subject to be gc-collect'ed. Even if it could contain pointers to data from the heap, the decision of collect yes/no can be made on whether the 'stack' data element in the thread structure is set or unset.

On a site note: data buffers provided by alloc's may have any content. How does gc_collect_root() deal with that?

dpgeorge, 2020-06-05:
> As far as I can tell, it is only used in mp_thread_gc_others() to tell, whether gc_collect_root() it performed on the threads stack.

Correct.

> But in the ESP32 port the stack buffer is provided by RTOS and is itself not subject to be gc-collect'ed.

Actually it will have root pointers in it.  On the stack of each thread (near the top) lives the `mp_state_thread_t` struct that contains the locals and globals dicts.  These must be reachable by the GC or they will be reclaimed.

> the decision of collect yes/no can be made on whether the 'stack' data element in the thread structure is set or unset.

Yes, I guess it could be, which could simplify the code.  But I guess this all originates from the unix port's thread handling which uses `ready` to tell if the thread can respond to a signal.

> data buffers provided by alloc's may have any content. How does gc_collect_root() deal with that?

It's a conservative GC and assumes any word it scans may be a pointer.

robert-hh, 2020-06-05:
Thanks. I should have known about the mp_state_thread_t, because I looked for that a few days ago. It is impressive how you recall each little detail.

closed 2020-06-05 by robert-hh