micropython/micropython #6114: File modification time is 0 for recent esp32 firmware

remibert, 2020-06-05:
Hello,

Try this:
>>> f=open("toto.txt","w")
>>> f.write("kkk")
3
>>> f.close()
>>> import os
>>> os.stat("toto.txt")[8]
0

Always 0 returned on esp32 internal flash, for sd card the right date is returned.

For esp32-idf3-20200605-unstable-v1.12-510-g1e6d18c91.bin,  esp32-idf3-20200529-unstable-v1.12-478-g1662a0b06.bin bug is present. 

The firmware esp32-idf3-20191220-v1.12.bin work correctly.

Regards.

robert-hh, 2020-06-05:
Do you use FAT or Littlefs for the internal file system?

remibert, 2020-06-05:
I don't understand the question, I upload the firmware and enter just the lines attached to the bug report nothing else.

I discovered this bug on esp32-cam, having taken the master branch of micropython and patched the sources to support the camera and build firmware. When checking on a standard esp32 with official firmware I found the same problem.

The date test on the sd was done on an esp32-cam, I use uos.mount (machine.SDCard (), "/ sd").
f = file.open ("/sd/toto.txt", "w") ...

And there I have no problem with sd on esp32-cam.

The models of esp32 are :
https://www.amazon.fr/gp/product/B07MMNKBPQ/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1 <https://www.amazon.fr/gp/product/B07MMNKBPQ/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1> 
https://www.amazon.fr/gp/product/B07VJ34N2Q/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1 <https://www.amazon.fr/gp/product/B07VJ34N2Q/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1> 

I wanted to correct this problem, but I don't found where the file date management is located in micropython or esp-idf sources.

tve, 2020-06-05:
A number of ports have switched to LittleFS (instead of FAT) as default filesystem type and the MicroPython interface to LittleFS does not associate any metadata (such as dates) with files. Hence they are reported as 0 in ilistdir.

tve, 2020-06-06:
Since you opened the ticket you can close it... ;-)

peterhinch, 2020-06-06:
> the MicroPython interface to LittleFS does not associate any metadata (such as dates) with files

Is there a plan to fix this? It has implications for the `rsync` feature of `rshell` which uses timestamps to determine which files to copy. Maybe some IDE's might be affected too?

tve, 2020-06-06:
Agreed, however, it might be smart to switch to using SHA1's for rsync instead of filesystem dates. The time it takes to calculate those is surprisingly small.

remibert, 2020-06-06:
Yes indeed I could also use this. But the problem with rsync is that 115200 baud is too limiting. I'm going to write a script to synchronize with ftp.

closed 2020-06-06 by remibert

davehylands, 2020-06-06:
A SHA1 would only tell you if the files were different, but not which one is newer. I suppose you could make an assumption that different means copy host->device

tve, 2020-06-09:
> A SHA1 would only tell you if the files were different, but not which one is newer. I suppose you could make an assumption that different means copy host->device

.. or have separate operations for syncing in one direction vs the other. If you can't set the modification time to the same value as the sync source, is bidir sync practical?

davehylands, 2020-06-09:
You really don't need something as complicated as a SHA1 even. A simple crc32 would be sufficient. The only thing you need to distinguish are 2 files which have the same name, same size but different content.

Having said that, it seems like crc32, sha1, md5 are not enabled by default on stm32, but sha256 does appear to be enabled by default (also on esp32). SHA1 is enabled by default on esp8266. I think crc32 is small enough that it can be implemented in python with a small amount of code.

tve, 2020-06-09:
> sha256 does appear to be enabled by default (also on esp32)

bingo!

To come back to the original issue, I do think it would be great to pass file times into LittleFS, but someone has to do the work...

peterhinch, 2020-06-10:
> is bidir sync practical?

Is there any real application?

> The only thing you need to distinguish are 2 files which have the same name, same size but different content.

The aim being to detect the common case of unchanged files and avoid the inefficiency of copying them (which is what happens at the moment).

I'm unclear how checksums would be used. Unless you store the checksums on the target device, you'd have to compute them on the fly. This means reading the entire file from the target, in which case there is no benefit over a bytewise comparison. Reading the entire file would be as slow as the current unconditional copy. Storing checksums on the device would work, but sounds like an rshell-specific kludge.

> I do think it would be great to pass file times into LittleFS

Yes.

remibert, 2020-06-10:
The date is not only useful for synchronization, it allows to know the last writing of the file. If you didn't have this on your computer, I think you would invent it. If adding a date is not possible on LittleFs, there is a serious risk of amputating its use.

dpgeorge, 2020-08-26:
File modification time for littlefs (v2) was implemented in 2acc087880de39d7e17abc9344b8d2faba3478dd