micropython/micropython #6116: BLE.active() is too heavey, MICROPY_EVENT_POLL_HOOK called too many times at mp_bluetooth_init()

jp-96, 2020-06-06:
bluetooth.BLE.active() is too heavery, waiting for 30 secondes.
MICROPY_EVENT_POLL_HOOK was called too many times for waiting active at mp_bluetooth_init().

simple code:
```
from bluetooth import BLE
b=BLE()
b.active(True)
```

MICROPY_EVENT_POLL_HOOK was called 79654 times on my device, esp32.

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4920
load:0x40078000,len:10592
ho 0 tail 12 room 4
load:0x40080400,len:5652
entry 0x400806c8
I (533) cpu_start: Pro cpu up.
I (534) cpu_start: Application information:
I (534) cpu_start: Compile time:     Jun  6 2020 15:42:35
I (537) cpu_start: ELF file SHA256:  0000000000000000...
I (543) cpu_start: ESP-IDF:          v3.3.2
I (548) cpu_start: Starting app cpu, entry point is 0x4008346c
I (0) cpu_start: App cpu up.
I (558) heap_init: Initializing. RAM available for dynamic allocation:
I (565) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (571) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (577) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (583) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (589) heap_init: At 3FFCA778 len 00015888 (86 KiB): DRAM
I (596) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (602) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (608) heap_init: At 4009DEA8 len 00002158 (8 KiB): IRAM
I (614) cpu_start: Pro cpu start user code
I (297) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
MicroPython v1.12-508-g621f40b12-dirty on 2020-06-06; ESP32 module with ESP32
Type "help()" for more information.
>>> from bluetooth import BLE
>>> b=BLE()
>>> b.active(True)
mp_bluetooth_init
mp_bluetooth_deinit
I (17840) BTDM_INIT: BT controller compile version [e989f20]
I (35983) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (36093) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
1 2 3 4 5 6 7 8 9 10 11 12  ...  79649 79650 79651 79652 79653 79654
MICROPY_EVENT_POLL_HOOK: 79654 times
mp_bluetooth_init: ready
True
>>> b.gap_advertise()
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=7 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=800 adv_itvl_max=800
>>> gap_event_cb: type=0
gap_event_cb: type=4
gap_event_cb: type=3
gap_event_cb: type=4
gap_event_cb: type=3
gap_event_cb: type=1
b.gap_advertise()
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=7 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=800 adv_itvl_max=800
>>>
```

jp-96, 2020-06-07:
It's lighter and faster.

Source code modified and using ESP-IDF v4.0.

1. call ble_svc_gatt_init()
2. call nimble_port_freertos_init(ble_host_task) after ble_svc_gap_init and ble_svc_gatt_init

diff:
```
 extmod/nimble/modbluetooth_nimble.c | 7 +++++--
 lib/berkeley-db-1.xx                | 0
 ports/esp32/nimble.c                | 2 +-
 3 files changed, 6 insertions(+), 3 deletions(-)

diff --git a/extmod/nimble/modbluetooth_nimble.c b/extmod/nimble/modbluetooth_nimble.c
index 05ec61177..035ec3d26 100644
--- a/extmod/nimble/modbluetooth_nimble.c
+++ b/extmod/nimble/modbluetooth_nimble.c
@@ -39,6 +39,7 @@
 #include "nimble/ble.h"
 #include "nimble/nimble_port.h"
 #include "services/gap/ble_svc_gap.h"
+#include "services/gatt/ble_svc_gatt.h"
 
 #ifndef MICROPY_PY_BLUETOOTH_DEFAULT_GAP_NAME
 #define MICROPY_PY_BLUETOOTH_DEFAULT_GAP_NAME "MPY NIMBLE"
@@ -271,8 +272,9 @@ int mp_bluetooth_init(void) {
     nimble_port_init();
     mp_bluetooth_nimble_port_postinit();
 
-    // By default, just register the default gap service.
+    // By default, just register the default gap/gatt service.
     ble_svc_gap_init();
+    ble_svc_gatt_init();
 
     mp_bluetooth_nimble_port_start();
 
@@ -466,8 +468,9 @@ int mp_bluetooth_gatts_register_service_begin(bool append) {
     // Reset the gatt characteristic value db.
     mp_bluetooth_gatts_db_reset(MP_STATE_PORT(bluetooth_nimble_root_pointers)->gatts_db);
 
-    // By default, just register the default gap service.
+    // By default, just register the default gap/gatt service.
     ble_svc_gap_init();
+    ble_svc_gatt_init();
 
     if (!append) {
         // Unref any previous service definitions.
diff --git a/lib/berkeley-db-1.xx b/lib/berkeley-db-1.xx
--- a/lib/berkeley-db-1.xx
+++ b/lib/berkeley-db-1.xx
@@ -1 +1 @@
-Subproject commit 35aaec4418ad78628a3b935885dd189d41ce779b
+Subproject commit 35aaec4418ad78628a3b935885dd189d41ce779b-dirty
diff --git a/ports/esp32/nimble.c b/ports/esp32/nimble.c
index e60e08bad..16829732c 100644
--- a/ports/esp32/nimble.c
+++ b/ports/esp32/nimble.c
@@ -44,7 +44,6 @@ void mp_bluetooth_nimble_port_preinit(void) {
 }
 
 void mp_bluetooth_nimble_port_postinit(void) {
-    nimble_port_freertos_init(ble_host_task);
 }
 
 void mp_bluetooth_nimble_port_deinit(void) {
@@ -52,6 +51,7 @@ void mp_bluetooth_nimble_port_deinit(void) {
 }
 
 void mp_bluetooth_nimble_port_start(void) {
+    nimble_port_freertos_init(ble_host_task);
 }
 
 #endif
``

dpgeorge, 2020-06-07:
This is very likely a duplicate of #5489 .  Note that the issue is apparent in examples from the IDF so it's a bug in the IDF itself.

closed 2020-06-11 by jp-96