micropython/micropython #6121: got error compiling upip.py

freeeflyer, 2020-06-07:
Hello All !

First my config : 
- Mac OS 10.13
- Python 3.8.2

All dependencies installed.

When executing : 
```
python3 ../../tools/makemanifest.py -o build-standard/frozen_content.c -v "MPY_DIR=../.." -v "MPY_LIB_DIR=../../../micropython-lib" -v "PORT_DIR=/Users/nico/code/micropython/ports/unix" -v "BOARD_DIR=" -b "build-standard" -f"-mcache-lookup-bc" manifest.py -v V=1
```
got following message : 
```
MPY upip.py
error compiling /Users/nico/code/micropython/tools/upip.py: b''
```

Thanks for any help..

freeeflyer, 2020-07-02:
no-one ?

robert-hh, 2020-07-02:
What happens if you go into the directory of upip.py and call <path_to_mpy_cross>/mpy-cross upip.py?

freeeflyer, 2020-07-02:
I git-pulled today, and now the compilation goes a bit further..
However it seems that the mpy-cross shoots a different erreur during make..

```
...
CC ../../extmod/uos_dupterm.c
CC ../../lib/embed/abort_.c
CC ../../lib/utils/printf.c
MPY upip.py
error compiling /Users/nico/code/micropython/tools/upip.py: b'dyld: malformed mach-o image: missing LC_DYSYMTAB\n'
make: *** [build-standard/frozen_content.c] Error 1
```

freeeflyer, 2020-07-02:
It relates to that #4455

closed 2020-07-02 by freeeflyer