micropython/micropython #6134: no output from tools/mpy-tool.py -d

jcw, 2020-06-09:
I'm trying to understand the structure of .mpy files. It's a complex format, but from the docs, it appears that `mpy-tool.py` should be able to decode these data structures, up to some point.

I'm not sure why, but when tested with a very simple .mpy file, it just returns. Has it gone stale?

```
$ cat hello.py
print('hello')
$ mpy-cross hello.py
$ ls -l hello.*
-rw-r--r--  1 jcw  staff  37 Jun  9 23:59 hello.mpy
-rw-r--r--  1 jcw  staff  15 Jun  9 23:20 hello.py
$ ~/code/git/micropython/tools/mpy-tool.py -d hello.mpy
$ ~/code/git/micropython/tools/mpy-tool.py -d hello.py
[...]
Exception: not a valid .mpy file
$ xxd hello.mpy
00000000: 4d05 021f 2048 080a 0007 1068 656c 6c6f  M... H.....hello
00000010: 2e70 7900 1100 7b10 0a68 656c 6c6f 3401  .py...{..hello4.
00000020: 5951 6300 00                             YQc..
```

dpgeorge, 2020-06-10:
The `-d` option to `mpy-tool.py` is not fully implemented.  For the time being the `-f` option on its own does a pretty good job, eg:

```
$ mpy-tool.py -f hello.mpy
...
// frozen bytecode for file hello.py, scope hello_<module>
STATIC const byte fun_data_hello__lt_module_gt_[18] = {
    0x08, 0x0a,
    MP_QSTR__lt_module_gt_ & 0xff, MP_QSTR__lt_module_gt_ >> 8,
    MP_QSTR_hello_dot_py & 0xff, MP_QSTR_hello_dot_py >> 8,
    0x00,
    0x11, MP_QSTR_print & 0xff, MP_QSTR_print >> 8, 
    0x10, MP_QSTR_hello & 0xff, MP_QSTR_hello >> 8, 
    0x34, 0x01, 
    0x59, 
    0x51, 
    0x63, 
};
const mp_raw_code_t raw_code_hello__lt_module_gt_ = {
    .kind = MP_CODE_BYTECODE,
    .scope_flags = 0x00,
    .n_pos_args = 0,
    .fun_data = fun_data_hello__lt_module_gt_,
    .const_table = NULL,
    #if MICROPY_PERSISTENT_CODE_SAVE
    .fun_data_len = 18,
    .n_obj = 0,
    .n_raw_code = 0,
    #if MICROPY_PY_SYS_SETTRACE
    .prelude = {
        .n_state = 2,
        .n_exc_stack = 0,
        .scope_flags = 0,
        .n_pos_args = 0,
        .n_kwonly_args = 0,
        .n_def_pos_args = 0,
        .qstr_block_name = MP_QSTR__lt_module_gt_,
        .qstr_source_file = MP_QSTR_hello_dot_py,
        .line_info = fun_data_hello__lt_module_gt_ + 0,
        .opcodes = fun_data_hello__lt_module_gt_ + 7,
    },
    .line_of_definition = 0,
    #endif
    #if MICROPY_EMIT_MACHINE_CODE
    .prelude_offset = 0,
    .n_qstr = 0,
    .qstr_link = NULL,
    #endif
    #endif
    #if MICROPY_EMIT_MACHINE_CODE
    .type_sig = 0,
    #endif
};

const char mp_frozen_mpy_names[] = {
"hello.py\0"
"\0"};
const mp_raw_code_t *const mp_frozen_mpy_content[] = {
    &raw_code_hello__lt_module_gt_,
};
```

jcw, 2020-06-10:
Ahhh, that does indeed help understand the frozen compiled-in version of a module. Thanks!

closed 2020-06-10 by jcw