micropython/micropython #6136: MIMXRT1010_EVK compiled fail based on the latest code.

yanyao12138, 2020-06-10:
Here is the log
> LINK build-MIMXRT1010_EVK/firmware.elf
arm-none-eabi-ld: build-MIMXRT1010_EVK/firmware.elf section `.stack' will not fit in region `m_data'
arm-none-eabi-ld: region m_data overflowed with stack and heap
arm-none-eabi-ld: region `m_data' overflowed by 680 bytes
Makefile:133: recipe for target 'build-MIMXRT1010_EVK/firmware.elf' failed
make: *** [build-MIMXRT1010_EVK/firmware.elf] Error 1

It seems due to the size limit of region m_data(Actually it is the DTCM memory), the stack and heap overflowed.
I propose to move some sections into the m_data2 region. Please refer to the following code.

``` __etext = .;    /* define a global symbol at end of code */
  __DATA_ROM = .; /* Symbol is used by startup for data initialization */

  .interrupts_ram :
  {
    . = ALIGN(4);
    __VECTOR_RAM__ = .;
    __interrupts_ram_start__ = .; /* Create a global symbol at data start */
    *(.m_interrupts_ram)     /* This is a user defined section */
    . += VECTOR_RAM_SIZE;
    . = ALIGN(4);
    __interrupts_ram_end__ = .; /* Define a global symbol at data end */
  } > m_data

  __VECTOR_RAM = DEFINED(__ram_vector_table__) ? __VECTOR_RAM__ : ORIGIN(m_interrupts);
  __RAM_VECTOR_TABLE_SIZE_BYTES = DEFINED(__ram_vector_table__) ? (__interrupts_ram_end__ - __interrupts_ram_start__) : 0x0;

  .data : AT(__DATA_ROM)
  {
    . = ALIGN(4);
    __DATA_RAM = .;
    __data_start__ = .;      /* create a global symbol at data start */
    *(m_usb_dma_init_data)
    *(.data)                 /* .data sections */
    *(.data*)                /* .data* sections */
    KEEP(*(.jcr*))
    . = ALIGN(4);
    __data_end__ = .;        /* define a global symbol at data end */
  } > m_data2
  __NDATA_ROM = __DATA_ROM + (__data_end__ - __data_start__);
  .ncache.init : AT(__NDATA_ROM)
  {
    __noncachedata_start__ = .;   /* create a global symbol at ncache data start */
    *(NonCacheable.init)
    . = ALIGN(4);
    __noncachedata_init_end__ = .;   /* create a global symbol at initialized ncache data end */
  } > m_data
  . = __noncachedata_init_end__;
  .ncache :
  {
    *(NonCacheable)
    . = ALIGN(4);
    __noncachedata_end__ = .;     /* define a global symbol at ncache data end */
  } > m_data

  __DATA_END = __NDATA_ROM + (__noncachedata_init_end__ - __noncachedata_start__);
  text_end = ORIGIN(m_text) + LENGTH(m_text);
  ASSERT(__DATA_END <= text_end, "region m_text overflowed with text and data")

  /* Uninitialized data section */
  .bss :
  {
    /* This is used by the startup in order to initialize the .bss section */
    . = ALIGN(4);
    __START_BSS = .;
    __bss_start__ = .;
    *(m_usb_dma_noninit_data)
    *(.bss)
    *(.bss*)
    *(COMMON)
    . = ALIGN(4);
    __bss_end__ = .;
    __END_BSS = .;
  } > m_data2

  .heap :
  {
    . = ALIGN(8);
    __end__ = .;
    PROVIDE(end = .);
    __HeapBase = .;
    . += HEAP_SIZE;
    __HeapLimit = .;
    __heap_limit = .; /* Add for _sbrk */
  } > m_data

  .stack :
  {
    . = ALIGN(8);
    . += STACK_SIZE;
  } > m_data

  /* Initializes stack on the end of block */
  __StackTop   = ORIGIN(m_data) + LENGTH(m_data);
  __StackLimit = __StackTop - STACK_SIZE;
  PROVIDE(__stack = __StackTop);

  .ARM.attributes 0 : { *(.ARM.attributes) }

  ASSERT(__StackLimit >= __HeapLimit, "region m_data overflowed with stack and heap")```

dpgeorge, 2020-06-10:
I guess this is due to the recent increase in buffer size for USB HS.

There is 1k of RAM in `m_data` which is dedicated to the heap, which is unused in MicroPython.  So this could be set to 0 via `__heap_size__ = 0;` in the `boards/MIMXRT1011.ld` file.

yanyao12138, 2020-06-10:
> I guess this is due to the recent increase in buffer size for USB HS.
> 
> There is 1k of RAM in `m_data` which is dedicated to the heap, which is unused in MicroPython. So this could be set to 0 via `__heap_size__ = 0;` in the `boards/MIMXRT1011.ld` file.

Thanks for your reply, It works! I will update the `boards/MIMXRT1011.ld` file.

yanyao12138, 2020-06-10:
Hi @dpgeorge, I have updated the `boards/MIMXRT1011.ld`, here is the [link](https://github.com/micropython/micropython/pull/6139)