micropython/micropython #6138: repl hang on esp32

TimonPeng, 2020-06-10:
unable to use repl and restart automatically all the time.

esp chip is ESP32D0WDQ6 (revision 1)

docker build:
```
FROM ubuntu:18.04

USER root

RUN apt-get update \
    && apt-get install -y git curl wget libncurses-dev flex bison gperf python python-pip python-setuptools \
    python-serial python-click python-cryptography python-future python-pyparsing python-pyelftools \
    cmake ninja-build ccache libffi-dev libssl-dev

WORKDIR /root

RUN mkdir esp && cd esp \
    && curl -O https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0.tar.gz \
    && tar -xzf xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0.tar.gz \
    && rm xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0.tar.gz

ENV PATH=/root/esp/xtensa-esp32-elf/bin:$PATH
ENV IDF_PATH=/root/esp-idf

RUN git clone https://github.com/espressif/esp-idf.git \
    && cd esp-idf \
    && git checkout 9e70825d1e1cbf7988cf36981774300066580ea7 \
    && git submodule update --init --recursive

RUN git clone https://github.com/micropython/micropython.git \
    && cd micropython/ \
    && git submodule update --init --recursive \
    && cd mpy-cross \
    && make -j4 PYTHON=python2

WORKDIR /root/micropython/ports/esp32

VOLUME ["/tmp"]

CMD make -j4 submodules PYTHON=python2 \
    && make -j4 PYTHON=python2 BOARD=GENERIC
```

```
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:4944
ho 0 tail 12 room 4
load:0x40078000,len:10600
ho 0 tail 12 room 4
load:0x40080400,len:5684
entry 0x400806bc
W (65) boot: PRO CPU has been reset by WDT.
W (65) boot: WDT reset info: PRO CPU PC=0x40062258
W (65) boot: WDT reset info: APP CPU PC=0x40085c68
I (490) cpu_start: Pro cpu up.
I (490) cpu_start: Application information:
I (490) cpu_start: Compile time:     Jun 10 2020 06:39:02
I (493) cpu_start: ELF file SHA256:  0000000000000000...
I (499) cpu_start: ESP-IDF:          v3.3.2
I (504) cpu_start: Starting app cpu, entry point is 0x40083a94
I (0) cpu_start: App cpu up.
I (514) heap_init: Initializing. RAM available for dynamic allocation:
I (521) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (527) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (533) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (539) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (546) heap_init: At 3FFCA9B0 len 00015650 (85 KiB): DRAM
I (552) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (558) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (564) heap_init: At 4009DE28 len 000021D8 (8 KiB): IRAM
I (571) cpu_start: Pro cpu start user code
I (253) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
```

tve, 2020-06-10:
How did you flash? It looks like you may have used 80Mhz flash speed which may not work? What do you see after the last line in your log above?

TimonPeng, 2020-06-10:
@tve 

```bash
esptool.py --chip esp32 --port /dev/cu.SLAB_USBtoUART erase_flash
esptool.py --port /dev/cu.SLAB_USBtoUART --baud 921600 write_flash --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 /tmp/firmware.bin
```

i change flash_freq to 40m, it's works now. why?

loop through the logs above

tve, 2020-06-11:
I don't quite know either why the 80Mhz flash mode doesn't work, but it doesn't...