micropython/micropython #6149: ESP32 : GPIO12 is dangerous! Can brick a board by using it

elliotwoods, 2020-06-15:
GPIO12 also serves a purpose defining the voltage supplied to the flash memory by the internal regulator (either 3.3V, 1.8V or 'OFF').
I've tried on a couple of boards here just opening GPIO12, and it has bricked both boards.

They are both now in a perpetual state of 

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
```

The board responds to esptool, but it cannot write or read to its flash memory
It is possible to use `espefuse.py` to disable this pin, wherein you must also define a voltage to use, and you can only perform this operation once (it cannot be undone).

It is not clear what voltage is used by the flash memory on the board which I'm using.
It's possible that the board uses 1.8V flash memory, and now that the regulator was set to 3.3V by accident by using GPIO12, the flash memory is now fried (I'm not sure how it detects the required voltage ordinarily). 

All together, it seems that using GPIO12 should come with some kind of warning since simply opening it can brick the board. Has anybody else had similar experiences here?

Elliot

elliotwoods, 2020-06-15:
I'm reading the code on the chip as 25Q128JVSQ which seems to be this one:
https://www.winbond.com/resource-files/w25q128jv%20revg%2004082019%20plus.pdf

Absolute maximum ratings:
[image]

Interestingly, it seems like 3.3V on the power pin wouldn't kill it, also interestingly it's a 128Mbit memory chip - which is much larger than the 4MB (32Mbit) which is reported from esptool / elsewhere (I presume this is a H/W limit for ESP32)

robert-hh, 2020-06-15:
As far as I understand the ESP32 data sheet, GPIO12 (or MTDI) is sampled during reset. That value determines the supply voltage of the external memory devices. Since it has an internal pull-down, the default is 3.3V, when left floating. If the pin is connected to an external pull-up, it will use 1.8 V.
And indeed, the 25Q128 is the 3.3V model. The 1.8V type would be labelled 25lq128..
So most likely it is a "software" problem. You can try to flash the firmware by setting the flash size manually to 4MB or 8MB.
The message
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
```
turns up even with a good flashed device

elliotwoods, 2020-06-16:
@robert-hh - thank you for the response

Firstly I should note that the pin input setting was effectively in the boot.py, so there is a small chance that every time it boots, it's working correctly, but then switching into a state where the flash is inaccessible and crashes (but the flash itself isn't damaged). 

Please note that all esptool commands fail to write/read flash memory, e.g.:

```
esptool --port COM6 --chip esp32 write_flash --flash_mode qio --flash_size 32m 0x1000 micropython\ports\esp32\build-GENERIC\firmware.bin

esptool erase_flash
```

Bot commands result in 
```
....
A fatal error occurred: Timed out waiting for packet content
```

calling `esptool read_flash 0x0 0x1000 output.bin` just hangs forever.

notably  `esptool flash_id` outputs:

```
esptool.py v2.8
Found 1 serial ports
Serial port COM6
Connecting........_
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 26MHz
MAC: 3c:71:bf:f0:4d:b4
Uploading stub...
Running stub...
Stub running...
Manufacturer: ff
Device: ffff
Detected flash size: Unknown
Hard resetting via RTS pin...
```

Which suggests the flash is completely inaccessible.
On the same board before the issue came it would read:

```
Manufacturer: ef
Device: 4018
Detected flash size: 16MB
```

The next step here is to slowly 'break' a new board to see when the issue arises.

elliotwoods, 2020-06-16:
Some progress here..

It actually seems to be an issue on another pin
calling `Pin(7, Pin.IN)` is what sends it into an infinite loop of 
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
```
and stops you from reading/writing to flash memory.

Thankfully a hard reset does fix the issue (which suggests that my earlier hypothesis about me having my pins initialized at boot might be what's causing the problem to persist).

Furthermore, holding down the PRG button on boot allows you to get past this issue (i.e. likely read status from the flash again (figuring out now how to get back to a working state through this on the boards which i thought i'd bricked).

So generally good news!
Since `GPIO7` is `SD_DATA_0`, which along with `GPIO8 / SD_DATA_1` are used to hook up external flash memory- this all makes sense.
i.e. you can't really use these pins for GPIO

closed 2020-06-16 by elliotwoods

elliotwoods, 2020-06-16:
btw - i closed the issue, but it seems like we might want to put some warnings into the documentation about GPIO pins which can cause adverse effects (sending people to the ESP32 datasheet is probably too much for some users, i.e. me before now)

dpgeorge, 2020-06-17:
> it seems like we might want to put some warnings into the documentation about GPIO pins which can cause adverse effects

There is already some info in the esp32 quick ref, it says: "Pins 6, 7, 8, 11, 16, and 17 are used for connecting the embedded flash, and are not recommended for other uses"

elliotwoods, 2020-06-17:
@dpgeorge  noted at : https://docs.micropython.org/en/latest/esp32/quickref.html#pins-and-gpio

Thank you everybody for being patient with me

dpgeorge, 2020-06-17:
Hardware is hard!

peterhinch, 2020-06-17:
I produced [this diagram](https://github.com/peterhinch/micropython-samples/blob/master/ESP32/ESP32-Devkit-C-pinout.pdf) some time ago: it is a pinout for the reference board with annotations about reserved pins etc. However there are a few differences. I have pins 9 and 10 listed as reserved for internal flash and pins 16 and 17 used for SPIRAM.

Much of this was taken from Espressif docs when the reference boards were first released so maybe some of it is out of date or incorrect.

someonewating, 2023-06-30:
Hi @elliotwoods , I think I have a highly similar problem to yours. Could you please share your method about the hard reset? I ran `esptool erase_flash` with other proper arguments and got the similar error message as yours, `A fatal error occurred: Packet content transfer stopped (received 8 bytes)`. Thanks.

elliotwoods, 2023-07-03:
@someonewating the solution for me was to not use those pins in my micropython programs, and to ensure that they definitely weren't being used in boot.py (otherwise I'd get very stuck).
By 'hard reset' I mean either unplugging/plugging the power back to the board, or reset with the reset switch.

sosi-deadeye, 2023-07-03:
I bricked an `ESP32` only once. It was a custom PCB with `MOSFets`. Accidentally, there was a connection between `+12V` and `+3.3V`. The interesting part is, it bricked only the Flash, but not the ESP32. I guess the flash _protected_ the ESP32. The ESP32 was still able to scream, that the flash was broken.

I was never able, with wrong/forbidden configurations of Pins, to brick the ESP32.
Even if an output has a short circuit, then the voltage drops to 0V and the output is current limited.

What I didn't know that a short-circuit could be detected via software. Just set the output and read the value. If output and value are different, then there is a short circuit to `GND` or `3.3V`.