micropython/micropython #6152: nrf: build is broken when SD=sxxx is specified

hoihu, 2020-06-15:
when compiling with softdevice specified, e.g. `make BOARD=pca10040 SD=s132`

the following error is thrown (except, there are more):
```
drivers/bluetooth/ble_drv.c:630:13: error: passing argument 2 of 'mp_raise_msg_varg' from incompatible pointer type [-Werror=incompatible-pointer-types]
  630 |             "Can not start advertisment. status: 0x" HEX2_FMT, (uint16_t)err_code);
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |             |
      |             char *
In file included from drivers/bluetooth/ble_drv.c:33:
../../py/runtime.h:164:84: note: expected 'mp_rom_error_text_t' {aka 'struct <anonymous> *'} but argument is of type 'char *'
  164 | NORETURN void mp_raise_msg_varg(const mp_obj_type_t *exc_type, mp_rom_error_text_t fmt, ...);
      |                                                                ~~~~~~~~~~~~~~~~~~~~^~~
```

I guess it was introduced when the move was done to the `MP_ERROR_TEXT` macros.

IMO there should be a buiild on travis with `SD=sxxx` to detect those build failures earlier.

dpgeorge, 2020-06-17:
> IMO there should be a buiild on travis with `SD=sxxx` to detect those build failures earlier.

Yes, agreed.

dpgeorge, 2020-06-17:
Build fixed by 8d71cc2e7da7a77aeca66bc8fac0bb0e28690344, Travis updated to build SD in cba3e25cb33007e4c0b1a0c476ecce7166fb089f

closed 2020-06-17 by dpgeorge