micropython/micropython #6154: MicroPython compatability issue with CPython

redradist, 2020-06-15:
Hi all,

I am wondering why `MicroPython` have so many inconsistencies with standard `CPython` ?

Do you plan to make it fully compatible with `CPython`  ?

We need it because it split ecosystem and decrease compatible between libraries for `MicroPython` and `CPython`

peterhinch, 2020-06-16:
See [this doc](https://github.com/micropython/micropython/wiki/Differences).

redradist, 2020-06-16:
@peterhinch 
> 
> See [this doc](https://github.com/micropython/micropython/wiki/Differences).

My point mostly about syntax compatibility:
1) Do not allow write `await` outside of `async def` function
2) Also why MRO is broken ... what if I want to use package library from PyPi that expect this behaviors ?
3) Also __del__ method missed, but when Garbage Collector delete object it could be called ...
For performance reason it could be stored in some flag for object and run-time then will check if flag present then method address should be searched
...

Also if some points is really decrease performance consider to add flag that will compile with 100% compatibility or with MicroController specific optimizations ...

What do you think about it ?

peterhinch, 2020-06-16:
I agree about point 1.

Fairly agnostic on MRO: I very rarely use MI and I've no idea what the cost in code size would be in fixing this.

The `__del__` method is contentious in CPython: the only guarantee is that it will be called before exit to the OS. Most firmware never exits; so even if MP implemented a CPython compatible version, it would be quite useless to firmware writers.

redradist, 2020-06-16:
@peterhinch 
> 
> I agree about point 1.

Cool, I hope it will be fixed as I saw an open issue for it ;)

> Fairly agnostic on MRO: I very rarely use MI and I've no idea what the cost in code size would be in fixing this.

Maybe you don't use MRO, but there are plenty of usable libraries that rely on it and therefor cannot be used in this implementation

Also one additional point ... How about of removing all syntax incompatibility in MicroPython https://docs.micropython.org/en/latest/genrst/syntax.html ?
Seems like there not many incompatibility ...

mattytrentini, 2020-06-25:
> plenty of usable libraries

Are you able to provide one or two examples so we can look into the issue? It's helpful having a 'real' example...

redradist, 2020-06-27:
> 
> 
> > plenty of usable libraries
> 
> Are you able to provide one or two examples so we can look into the issue? It's helpful having a 'real' example...

@mattytrentini I do not have any particular example ... It is general question that if it possible to do ... ;)

dpgeorge, 2020-06-28:
> I am wondering why `MicroPython` have so many inconsistencies with standard `CPython` ?

Considering the size of the language (excluding the libraries) I'd say MicroPython is highly consistent with CPython.  It's mainly in the corner cases that it can differ, due to MicroPython being a completely independent implementation.  And it has inconsistencies because it needs to run on very constrained systems.  The aim is to be able to run in such places, which is more important than being 100% consistent.

> Do you plan to make it fully compatible with CPython ?

No, because then it won't be able to run on a microcontroller or in constrained systems.  If you need 100% CPython compatibility then you need to use CPython.

> We need it because it split ecosystem and decrease compatible between libraries for MicroPython and CPython

It's possible to write a lot of code which runs the same under both MicroPython and CPython.

> Also one additional point ... How about of removing all syntax incompatibility in MicroPython 
> Seems like there not many incompatibility ...

It's not possible to do this, since adding all unicode names will increase the size of the system by a huge amount (megabytes).

>  Also why MRO is broken

Because features are implemented/developed on an as-needed basis, and fully compliant MRO is low on that list.  See #6077 and #525 for existing issues discussing MRO.

redradist, 2020-06-30:
> 
> 
> > Also one additional point ... How about of removing all syntax incompatibility in MicroPython
> > Seems like there not many incompatibility ...
> 
> It's not possible to do this, since adding all unicode names will increase the size of the system by a huge amount (megabytes).

@dpgeorge Seems like it is good answer that could be add to syntax differences for clarification, because it is not obvious

dpgeorge, 2020-07-01:
> Seems like it is a good answer that could be add to syntax differences page for clarification

Agreed.  The documentation of those differences could do with some updating and tidying.