micropython/micropython #6162: machine.SDCard freq-argument not recognized?

siochs, 2020-06-17:
The [documentation](https://docs.micropython.org/en/latest/library/machine.SDCard.html#machine.SDCard) states:
`classmachine.SDCard(slot=1, width=1, cd=None, wp=None, sck=None, miso=None, mosi=None, cs=None, freq=20000000)`

Now, what am I doing wrong?

```
>>> import machine, uos
>>> machine.SDCard(slot=2, width=1, freq=20000000)
Traceback (most recent call last):
File "<stdin>", line 1, in <module>
TypeError: extra keyword arguments given

>>> machine.SDCard(slot=2, width=1, cd=None, wp=None, sck=None, miso=None, mosi=None, cs=None, freq=20000000)
Traceback (most recent call last):
File "<stdin>", line 1, in <module>
TypeError: extra keyword arguments given

>>> machine.SDCard(2, 1, None, None, None, None, None, None, 20000000)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: extra positional arguments given
>>> machine.SDCard(2, 1, 10000000)
```
Where can I find the class definition?

siochs, 2020-06-17:
Oh, it seems that the freq-arguments only works with the lastest unstable betas. Maybe that should be noted somewhere in the docs.

dpgeorge, 2020-06-18:
> it seems that the freq-arguments only works with the lastest unstable betas. Maybe that should be noted somewhere in the docs.

That's right.  What the docs need is a banner at the top to note development version, rather than stable version.

siochs, 2020-07-16:
Good for me. Issue can be closed. Thank you guys.

closed 2020-07-16 by dpgeorge