micropython/micropython #6166: ESP32-spiram - UART panic error

pidou46, 2020-06-18:
Hello,

I get a panic error when trying to initialize UART port, with ESP32-spiram firmware.

I have tried with the stable V1.12 and the latest:

`[quote]MicroPython v1.12-553-gc521c178e on 2020-06-18; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> 
>>> 
>>> from machine import UART, PIn
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: can't import name PIn
>>> from machine import UART, Pin
>>> uart=UART(2, 9600)
>>> 
>>> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400dfb48  PS      : 0x00060d30  A0      : 0x800d2dc9  A1      : 0x3ffce8b0  
A2      : 0x3f8177c0  A3      : 0x000000b1  A4      : 0x3ffce8d0  A5      : 0xff1d0000  
A6      : 0x00000001  A7      : 0x000000ff  A8      : 0x00000019  A9      : 0x3ffce890  
A10     : 0xff1d0000  A11     : 0x00000001  A12     : 0x00000001  A13     : 0x3ffc2338  
A14     : 0x00000003  A15     : 0x3ffce990  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xff1d001c  LBEG    : 0x400d2fe0  LEND    : 0x400d3038  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x400dfb48:0x3ffce8b0 0x400d2dc6:0x3ffce8d0 0x400f74ec:0x3ffce910 0x400d303b:0x3ffce930 0x400d3364:0x3ffce950 0x400d2974:0x3ffce970 0x400d621d:0x3ffce990 0x400d62e1:0x3ffce9b0 0x400d6c09:0x3ffce9d0 0x4010741e:0x3ffcea50 0x40107531:0x3ffceae0 0x400f739d:0x3ffceb10 0x4009571e:0x3ffceb40

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5008
ho 0 tail 12 room 4
load:0x40078000,len:11608
ho 0 tail 12 room 4
load:0x40080400,len:6120
entry 0x400806bc
I (615) psram: This chip is ESP32-D0WD
I (615) spiram: Found 64MBit SPI RAM device
I (615) spiram: SPI RAM mode: flash 40m sram 40m
I (618) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (625) cpu_start: Pro cpu up.
I (629) cpu_start: Application information:
I (634) cpu_start: Compile time:     Jun 18 2020 00:56:09
I (640) cpu_start: ELF file SHA256:  0000000000000000...
I (646) cpu_start: ESP-IDF:          v3.3.2
I (651) cpu_start: Starting app cpu, entry point is 0x40083574
I (641) cpu_start: App cpu up.
I (1526) spiram: SPI SRAM memory test OK
I (1527) heap_init: Initializing. RAM available for dynamic allocation:
I (1527) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1533) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1539) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1546) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (1552) heap_init: At 3FFCABA8 len 00015458 (85 KiB): DRAM
I (1558) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1564) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1571) heap_init: At 4009A528 len 00005AD8 (22 KiB): IRAM
I (1577) cpu_start: Pro cpu start user code
I (254) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
MicroPython v1.12-553-gc521c178e on 2020-06-18; ESP32 module (spiram) with ESP32
Type "help()" for more information.[/quote]`

The error can rise immediately or after some delay.

pidou46, 2020-06-18:
The error doesn't rise with the non spiram firmware

robert-hh, 2020-06-18:
You have to specify the tx and rx pins. The default pins are used by the RAM.

pidou46, 2020-06-19:
On the sample I got from doc or forum, tx and rx pins are not set by the constructor, but after by calling: uart.init

Can I set the pins with the constructor?
I've tried, but it hang like this:

`>>> from machine import UART
>>> test=UART(2,9600,tx=1,rx=3)
I (4692

'

pidou46, 2020-06-19:
I've try to add some parameters:

uart=UART(2,baudrate=9600, bits=8, parity=None, stop=1, rx=3, tx=1)

But it still hang.

robert-hh, 2020-06-19:
Yes, you can use all init parameters in the constructor
but: Pin 1 and 3 are used for UART0, the one used for REPL. So better go for other pins. like 12 and 13. or 4 and 5. Please note, that pins 34-39 can only be used for input. You can use them for rx, but not tx

pidou46, 2020-06-19:
Thank you,
Does it worth adding a UART subsection in ESP32 doc section to give this information ?

robert-hh, 2020-06-19:
Or change the defaults. As far as I recall, that was mentioned once but not done.

pidou46, 2020-06-25:
Changing the defaults would be prevent noobs like me to face this error and open a new issue.

But an addition to the doc would give all the necessary informations.

pidou46, 2020-06-25:
Forget what I've just wrote, the information is given in the Pin section:

`Notes:

    Pins 1 and 3 are REPL UART TX and RX respectively

    Pins 6, 7, 8, 11, 16, and 17 are used for connecting the embedded flash, and are not recommended for other uses

    Pins 34-39 are input only, and also do not have internal pull-up resistors`