micropython/micropython #6170: Supporting insertion-order preservation property of dicts (compat with Python 3.7+)

dpgeorge, 2020-06-18:
Python 3.7 made dicts ordered by default, see https://docs.python.org/3/whatsnew/3.7.html

Since it's part of the Python spec, and there are situations where this feature is useful, it makes sense to consider how MicroPython could support it.  It may be that it's not realistic to have such a feature in MicroPython by default, maybe in the end it's just a build option for those who need it.

The problem is that implementing it efficiently (retaining O(1) lookup) requires more memory for the dict implementation than is currently used.  The alternative is to make lookups O(N) which is proposed in #5323 (see that PR for further discussion).

Another alternative might be to just have ordered dicts only in certain places.  There are currently requests for it in the following places:
- class members / locals-dict: #6130 
- json loading of dicts: #6135

stinos, 2020-06-18:
> Another alternative might be to just have ordered dicts only in certain places

Wouldn't that get confusing / error-prone? Suppose you get 2 ordered dicts from json then need to merge them, or copy one of them or so, that copy or merged dict(**a, **b) would not retain order?

andrewleech, 2020-06-18:
Note: All dict's are ordered by default in cpython 3.6 with the new dict implementation they brought in, though the ordering was a side effect at that point. They only made it part of the official spec in 3.7

@stinos the specific area orderings I've proposed in the PR's above has been an effort to preserve order in areas where it cannot be manually controlled any other way to support use cases of my project. They were suggested as an alternative to the compile switch to change dict's globally (#5323) due to the performance issues when ordered dicts get larger.  
With those two cases, yes, if you convert / merge the returned ordered dicts into a regular dict the ordering is lost. You could still copy/merge them into an OrderedDict if so desired however.

dpgeorge, 2020-06-19:
See #6173 for an implementation of order preserving dicts that has O(1) lookup, but uses more RAM.

dpgeorge, 2020-06-19:
> > Another alternative might be to just have ordered dicts only in certain places
> 
> Wouldn't that get confusing / error-prone?

Yes it probably will.  Like, it might not be obvious which places do use ordered dicts and which not, and you might rely on it when its not there (and simple testing can't tell because non-ordered dicts are sometimes ordered by accident).

Having various options to enabled ordered-by-default in various places will also lead to more confusion around the compile-time configuration options.  So if this approach were taken then I'd suggest having just one config option to enabled it in all the various places.

But I would lean towards just doing it properly like in #6173, and make all dicts ordered, rather than having it done in an ad-hoc way in various places.

stinos, 2020-06-20:
>  having just one config option to enabled it in all the various places

+1

Crsarmv7l, 2024-01-07:
> > Another alternative might be to just have ordered dicts only in certain places
> 
> Wouldn't that get confusing / error-prone? Suppose you get 2 ordered dicts from json then need to merge them, or copy one of them or so, that copy or merged dict(**a, **b) would not retain order?

I am late to the game, but it was very confusing and error prone for me to grab values from a dict on my computer with the order preserved in the standard dict, then unknowingly experiencing the order changing on my ESP32 (in my circumstance the order does matter).

Sure I fixed it with collection and an ordered dict, but aside from the available libs difference between computer/MCU, this is the first real difference in operations I have seen