micropython/micropython #6175: Lookup of operator builtins can fail with multiple inheritance

nickovs, 2020-06-19:
When looking up a method that implement an operator (such as `__int__()` or `__lshift__()`) on an object who's class uses multiple inheritance, and an earlier base classes is a native type that does not implement that operator, then an implementation of the required method will not be found in a later base class that does implement it.

Take for instance the following example:
```
class Foo:
    def __int__(self):
        return 123

class Bar(Foo, list):
    pass

class Baz(list, Foo):
   pass

print(int(Bar([])))
print(int(Baz([])))
```

On C Python this prints the expected:
```
123
123
```
On Micropython the second call raises a `TypeError`:
```
123
Traceback (most recent call last):
  File "<stdin>", line 12, in <module>
TypeError: can't convert list to int
```

This is because `mp_obj_class_lookup()` checks to see if a native type has the necessary `unary_op` or `binary_op` function pointer but does not check to see if the specific operator code is implemented. If a native type if found that has the required operator function pointer the search for an implementation stops there, even if it doesn't implement the desired operator code, preventing a later, usable implementation from being found.

nickovs, 2020-06-19:
For completeness, here is a test for the binary operator case:
```
class Foo:
    def __lshift__(self, n):
        return "Shifty: {}".format(n)

class Bar(Foo, list):
    pass

class Baz(list, Foo):
    pass

print(Bar() << 3)
print(Baz() << 3)
```

dpgeorge, 2020-06-21:
Thanks for the report.  Inheriting from a native type is not fully supported, and it looks like this is another (uncommon?) case where it doesn't work.

nickovs, 2020-06-21:
I agree that this is likely a pretty uncommon case. It only happens in _a subset of_ the scenarios where you are inheriting from a native type *and* that native type is _not_ at the end of the method resolution order.

dpgeorge, 2020-06-21:
At least it fails with an error, rather than silently doing the wrong thing.

nickovs, 2020-06-21:
True. Given the known limitations associated both multiple inheritance and subclassing native types it probably makes sense to add a note in the MicroPython differences from CPython" documentation and leave it at that. I can't see a way to fix this without adding either significant complexity or significant space consumption.