micropython/micropython #6183: GeneratorExit is not thrown when generator is interrupted

jancespivo, 2020-06-23:
Hi,
the code:
```python
def generator():
    try:
        for i in range(10):
            yield i
    except GeneratorExit:
        print('GeneratorExit')

for i in generator():
    if i > 3:
        break
    print(i)
```

In CPython prints 
```
0
1
2
3
GeneratorExit
```

In MicroPython:
```
0
1
2
3
```

Thanks for your work!

dpgeorge, 2020-06-23:
It actually cancels the generator (and prints "GeneratorExit") when the generator is reclaimed by the memory system.  In CPython this is due to reference counting, so happens as soon as the for-loop ends.  But in MicroPython it never happens, because of the way the garbage collector works.  This is still technically within the Python spec because Python does not require reference counting, and in a garbage collected system it's not guaranteed that memory is reclaimed, or when it is reclaimed.

In what way does your code rely on this behaviour?

jancespivo, 2020-06-23:
Thanks for your reply. I see the reason, why it is not raised. Our use-case is to scan via Bluetooth module indefinitely, yield the result one by one, and if a right device is found it is needed to stop the scan. AFAIK the `GeneratorExit` was introduced in https://www.python.org/dev/peps/pep-0342

Btw the below is a more generic version, which doesn't work either.

```python
def generator():
    try:
        for i in range(10):
            yield i
    finally:
        print('Finally')

for i in generator():
    if i > 3:
        break
    print(i)
```

It should print 
```
0
1
2
3
Finally
```

dpgeorge, 2020-06-24:
From PEP 342, MicroPython supports all parts of the specification summary https://www.python.org/dev/peps/pep-0342/#specification-summary except for part 5 which relates to garbage collection.

To make it work you should be able to call `gen.close()` explicitly, like this:
```python
def generator():
    try:
        for i in range(10):
            yield i
    finally:
        print('Finally')

g = generator()
for i in g:
    if i > 3:
        break
    print(i)
g.close()
```

jancespivo, 2020-06-24:
Thanks for the clarification. However, please imagine the code:

```python
def bluetooth_scan():
    gap_scan(0) # scan indefinitely
    try:
        while scan_results:  # deque with results
            yield scan_results.popleft()
    finally:
        gap_scan(None) # stop scan    
```

Ideally the usage would be:
```python
def find_device(name):
    for device in bluetooth_scan():
        if device.name == name:
            return device
```

But because finalization is not executed, the code has to be:
```python
def find_device(name):
    bluetooth_scan_g = bluetooth_scan()
    for device in bluetooth_scan_g:
        if device.name == name:
            bluetooth_scan_g.close()
            return device
```
IMO it isn't an obvious way how to use the generator and it makes the code more complex.

Is there any plan to make it work? I think the `GeneratorExit` isn't so important but it looks like a bug if `finally` section doesn't work either (because of `finally` semantics). Thanks!

peterhinch, 2020-06-24:
Similar ends may be achieved with the generator **send** or **throw** methods.

dpgeorge, 2020-06-25:
> But because finalization is not executed, the code has to be:
> ... 
> IMO it isn't an obvious way how to use the generator and it makes the code more complex.

In general it's not a good idea to rely on a finaliser being called, because it could be that something else refers to the object and the finaliser is not called until much later on in the code (even in CPython).  In MicroPython this is definitely true, that finalisers (if they are called) are not called until a garbage collection cycle.

> Is there any plan to make it work?

It's impossible to make it work with just a garbage collector, it needs reference counting (unless the GC is called every opcode, which will severely impact performance of the VM).  And since the Python spec does not require reference counting the exact time at which the generator is finished is not properly defined within the language spec.  So I suggest to rewrite the code not to rely on this automatic closing of the generator.

jancespivo, 2020-06-26:
Thank you for the exhaustive answers! I've found the problem really exists also in cpython and there is (deffered) PEP about it https://www.python.org/dev/peps/pep-0533/ I think the issue can be closed (at least for me) for now and I'll try to find another solution.

dpgeorge, 2020-06-26:
Ah, I didn't know about that PEP 533, thanks for linking to it, very interesting.

I'll close this issue then.

closed 2020-06-26 by dpgeorge