micropython/micropython #6185: [BLE] Not able to subscribe to notification from a peripheral device

p1ng0o, 2020-06-24:
As a Central Role (GATT Client), and once connected, I am not able to subscribe to notify event, for a caracteristic that only implements FLAG_NOTIFY.

 The _IRQ_GATTC_NOTIFY event never gets fired.

Port: ESP32
firmware: esp32spiram-idf3-20200624-unstable-v1.12-572-gb4dc4c5b9.bin

dpgeorge, 2020-06-24:
Can you please post the code that doesn't work so that the issue can be reproduced?

p1ng0o, 2020-06-24:
Sorry for the lack of info. I think without context, the whole code will mean nothing :(
I will try to explain in more details.

If I am right, I have to enable notification inside the 0x2902 descriptor of the caracteristics, in order to start receiving ble notifications.

I am doing it by making a write on the client characteristic configuration descriptior:
`ble.gattc_write(connexion_handle, 49, b'\x00\x01', 1)`
Where 49 is the 0x2902 descriptor of the caracteristic.

I am processing the async
`    elif event == 17:
        conn_handle, value_handle, status = data
        print("Write done", status)
`
Which output "Write done 384". And 384 doesn't seems to be a valid nimble return status code.

In an other hand, I am able to read this descriptor: 0x00 0x00

dpgeorge, 2020-06-25:
> Why the bytes orders are reversed vs spec

I think it should be little endian, and the notify bit is bit 0, so the value to write is 0x0001 = b`\x01\x00`.

> How to look after the right error codes ?

384 = 0x180, and 0x100 is the base error code for `ATT_ERR` (see `BLE_HS_ERR_ATT_BASE`).  But 0x80 doesn't seem to be a valid error code so I'm not sure...

jimmo, 2020-06-29:
> ble.gattc_write(connexion_handle, 49, b'\x00\x01', 1)

> I think it should be little endian, and the notify bit is bit 0, so the value to write is 0x0001 = b`\x01\x00`.

Yep, exactly this.

Use struct.pack when working with BLE char/desc values to avoid issues around byte order.

```
_NOTIFY_ENABLE = const(1)
_INDICATE_ENABLE = const(2)

ble.gattc_write(conn_handle, desc_handle, struct.pack('<h', _NOTIFY_ENABLE), 1)`
```

closed 2020-06-29 by p1ng0o