micropython/micropython #6186: Pyboard D build does not include btree

peterhinch, 2020-06-24:
The MICROPY_PY_BTREE option is specified for ESP8266 and ESP32, but not for the Pyboard D. Is this intentional?

dpgeorge, 2020-06-25:
The main reason is due to the licensing conditions of the berkeley-db code, one of the clauses is:
```
 * 3. All advertising materials mentioning features or use of this software
 *    must display the following acknowledgement:
 *      This product includes software developed by the University of
 *      California, Berkeley and its contributors.
```
and that puts constraints on marketing of pyboard.

But from what I understand this clause was removed from all Berkeley code, so would be good to make that change to the license in this library code, then it could be included in builds without a problem.

In the meantime, `btree` can be built as a native .mpy file:
```
$ cd examples/natmod/btree
$ make ARCH=armv7m
```
Then copy `btree_armv7m.mpy` to the board as `btree.mpy`.

peterhinch, 2020-06-25:
This is another example of https://github.com/micropython/micropython/issues/5564. In response to a forum query I suggested **btree**, only for the user to discover that it wasn't available on his PBD. Embarrassing.

We need a quick way to check whether feature X is available on platform Y. The unappetising choice is to trawl the build system or plug in a board (if we have one) and try it.

fxwiegand, 2023-04-22:
> This is another example of #5564. In response to a forum query I suggested **btree**, only for the user to discover that it wasn't available on his PBD. Embarrassing.
>
> We need a quick way to check [...]

Almost three years later and this is still a problem. Took me quite some time to find out that `btree` isn't available on the pico. Not to mention that its a proper adventure to create a firmware that includes it only finding out that it's not working properly on the pico.

robert-hh, 2023-04-22:
btree was added to the esp8266 port after a poll in the user community. I never understood why, because the ESP8266 is very limited in resources, resulting in a limited implementation. Did you try if it works different on the Pico than on the ESP8266 or ESP32?

fxwiegand, 2023-04-22:
I haven't tried it on any of the ESP ones. Only on the pico compared to the unix port. I have talked to @iabdalkader on the micropython discord who also mentioned that there are problems with the btree module on the pico.

iabdalkader, 2023-04-22:
I manged to build btree for `rp2` (I get undefined references to `__bt_sync` etc...) but it still crashes/freezes on the `rp2` when flushing the db. It might be an architecture-specific issue, misalignment or something like that. Note I see the same errors when building `esp32`/GENERIC_S3 (`btree` is enabled there by default):

```C
micropython/extmod/modbtree.c: In function 'btree_flush':
micropython/extmod/modbtree.c:94:33: error: implicit declaration of function '__bt_sync'; did you mean '__bt_open'? [-Werror=implicit-function-declaration]
     return MP_OBJ_NEW_SMALL_INT(__bt_sync(self->db, 0));
```

Also note the native module does Not build for `armv6` either:

```bash
make ARCH=armv6m
GEN build/btree_armv6m.config.h
....
LINK build/btree_c.o
LinkError: build/micropython/lib/berkeley-db-1.xx/btree/bt_open.o: undefined symbol: __aeabi_uidiv
make: *** [../../../py/dynruntime.mk:150: build/btree_armv6m.native.mpy] Error 1
```

darconeous, 2023-10-08:
Anyone have any ideas for getting past the `bt_open.o: undefined symbol: __aeabi_uidiv` error?

vshymanskyy, 2024-09-13:
@darconeous please check out this PR: https://github.com/micropython/micropython/pull/15838
This should resolve the `undefined symbol: __aeabi_uidiv` problem