micropython/micropython #6190: oofatfs/ff.c compilation error

xidameng, 2020-06-25:
While building firmware for a third party port, this error occur

`../../lib/oofatfs/ff.c:559:37: error: pasting "TBL_CT" and "(" does not give a valid preprocessing token
  559 | static const BYTE ExCvt[] = MKCVTBL(TBL_CT, FF_CODE_PAGE);
      |                                     ^~~~~~
../../lib/oofatfs/ff.c:422:26: note: in definition of macro 'MERGE_2STR'
  422 | #define MERGE_2STR(a, b) a ## b
      |                          ^
../../lib/oofatfs/ff.c:559:29: note: in expansion of macro 'MKCVTBL'
  559 | static const BYTE ExCvt[] = MKCVTBL(TBL_CT, FF_CODE_PAGE);
      |                             ^~~~~~~
make: *** [../../py/mkrules.mk:75: build/genhdr/qstr.i.last] Error 1
`
Has anyone reported this already or I did something wrong?

xidameng, 2020-06-25:
I fixed this issue by replacing MKCVTBL with MERGE_2STR

Did anyone faced the same issue before?

dpgeorge, 2020-06-26:
As far as I know, this issue has not been seen before.  What compiler and preprocessor are you using?

xidameng, 2020-06-26:
> As far as I know, this issue has not been seen before. What compiler and preprocessor are you using?

Hi, I was using GCC v4.8.3, but also tried GCC v9.2.1, the error persist.

Interesting finding -- after replacing "MKCVTBL" with "MERGE_2STR", it worked, but with another error saying "TBL_CTFF_CODE_PAGE" is not declared in the scope.

I figured the second error was mostly likely caused by preprocessor not expanding the "FF_CODE_PAGE" with its actual definition (in my case, its 437), but I not sure why it happened.

dpgeorge, 2020-06-26:
Make sure `FF_CODE_PAGE` is defined via your configuration file, you need something like `CFLAGS += -DFFCONF_H=\"myffconf.h\"`

xidameng, 2020-06-26:
> Make sure `FF_CODE_PAGE` is defined via your configuration file, you need something like `CFLAGS += -DFFCONF_H=\"myffconf.h\"`

Thanks for the reply, tried, no luck 
Compiler still throw error of the same

dpgeorge, 2020-06-26:
There are lots of example ports that do build correctly with oofatfs, so best to look at how they work and copy what they do.

xidameng, 2020-06-26:
Thanks, I will take a look, seems it is not a issue for other case, I will close this for now

closed 2020-06-26 by xidameng