micropython/micropython #6191: list.index() with mixed bytes and string objects in the list gives a weird warning.

gmos, 2020-06-25:
list.index() with mixed bytes and string objects in the list gives a weird warning.  Oddly enough list.index() does return the right value in all cases.  Just the warning should go.

>>> [b'b','s'].index('s')
Warning: Comparison between bytes and str
1
>>> [b'b','s'].index(b'b')
0

and if we put a string upfront:

>>> ['s',b'b'].index('s')
0
>>> ['s',b'b'].index(b'b')
Warning: Comparison between bytes and str
1

dpgeorge, 2020-06-26:
Thanks for the reports.  Actually, it looks like CPython also gives the same warnings for the same code.  But in CPython 3.7 the warnings are disabled by default, and to see the warnings there, use `python3 -b test.py`.

But there's a more serious version of the problem in MicroPython which doesn't seem to show up on CPython, using a set/dict, eg:
```python
print('s0' in {'s0', 's1', b'b0', b'b1'})
print(b'b0' in {'s0', 's1', b'b0', b'b1'})
```

I'm not exactly sure how CPython deals with this case, why it doesn't warn.

Perhaps MicroPython should disable these by default (on esp32 and esp8266 builds at least), and provide the '-b' option on the unix/windows ports.

gmos, 2020-06-26:
Tried on cpython.  No problem there, but I had missed the warnings-off part.