micropython/micropython #6194: VFS not loaded onto NRF chip

someusername-123456, 2020-06-26:
I'm having trouble loading the VFS onto my custom board using a nrf52832. The board is not using an SD card and the only connections are UART lines (Rx and Tx) and 2 LEDs. 

Within a Virtual Machine on windows I'm running Linux and have completed the following steps:
- I've altered the UART and LED pins in the micropython/ports/nrf/boards/pca10040/mpconfigboard.h to match my configuration 
- I've set MICROPY_VFS in mpconfigport.h 
- I've compiled using make BOARD=pca10040 MICROPY_VFS_FAT=1, this compiles successfully and generates the following hex 
[image]

On windows
- On command line I can run the command "ampy -p COM9 run blink.py" which just blinks the on board LEDs
However when I run the command
"ampy -p COM9 ls" 
It just returns with nothing and the command line prompt appears. This leads me to believe that the VFS has not been installed correctly on the nrf chip. 

I am able to open a python terminal via PuTTY and when I type the command "os.listdir()" it returns empty 
[image]

Have I missed a step when compiling perhaps?
Any help would be greatly appreciated

dpgeorge, 2020-06-26:
What does `dir(os)` report?  You may need to explicitly format the filesystem.  Or it could be that it's there and working and you just need to make some files, eg `f=open('test.txt', 'w'); f.write('test'); f.close()`

someusername-123456, 2020-06-26:
This is what dir('os') reports. also I get an OSError when creating files 
[image]

How would I go about explicitly formatting the filesystem?

glennrub, 2020-06-26:
Today, `microbitfs` (MBFS) is default internal flash filesystem for all nrf-targets. Setting `MICROPY_VFS (1)`, will exclude MBFS ([moduos.c L142](https://github.com/micropython/micropython/blob/master/ports/nrf/modules/uos/moduos.c#L142) and [moduos.c L156](https://github.com/micropython/micropython/blob/master/ports/nrf/modules/uos/moduos.c#L156)). This means that you might have VFS and FATFS compiled in, but it is only usable for manually mounted partitions, like an external SD card. In practice this means that you do not have any internal storage, nor any mounts unless you also mount and SD card with FATFS. The MBFS integration is not compatible with running `ampy` and similar tools.

I recommend that you look at PR #5472, where the internal flash filesystem can be set to `LittleFS`, which should also work better with `ampy`. I would not recommend using `FATFS` for internal storage, but rather `LittleFS`. I noticed the PR has conflicts, i'll fix those in a few days.

someusername-123456, 2020-07-03:
Thanks for that Glen. After looking at PR #5472 I'm still having a bit of trouble with LittleFS now. I've followed the steps to enable LFS namely: 

- I've set `MICROPY_VFS` in mpconfigport.h
- I've compiled using make BOARD=pca10040 MICROPY_VFS_LFS=2 
However this throws the following error: `INK build-pca10040/firmware.elf
/tmp/ccriKTiK.ltrans3.ltrans.o: In function `os_sync.lto_priv.327':
/home/mira/micropython/ports/nrf/modules/uos/moduos.c:90: undefined reference to `disk_ioctl'
/tmp/ccriKTiK.ltrans2.ltrans.o:(.rodata+0x1840): undefined reference to `mp_fat_vfs_type'
collect2: error: ld returned 1 exit status
Makefile:447: recipe for target 'build-pca10040/firmware.elf' failed
make: *** [build-pca10040/firmware.elf] Error 1`

-If I compile with make BOARD=pca10040 MICROPY_VFS_LFS=2 MICROPY_VFS_FAT=1
this compilation succeeds, however when I flash and test it out I'm still receiving the same issue where I can't create directories as it throws an OSError. 

Is there any steps that I'm missing?

glennrub, 2020-07-03:
The `MICROPY_VFS_LFS=2` seems a bit wrong, last part of the parameter should be `LFS2=1` and not `LFS=2`. If i recall correctly, i have seen these errors before when having `VFS` without any `FS` selected. 

Also, add the two following to lines to your `boards/pca10040/mpconfigboard.h`:
```
#define MICROPY_VFS                 (1)
#define MICROPY_HW_ENABLE_STORAGE   (1)
#define MICROPY_HW_BDEV_READBLOCKS_EXT (1)
```

> Is there any steps that I'm missing?

The missing instruction is probably that you might have to freeze the "_mkfs.py" (located in the `freeze` folder). I will change this part a bit. But for now, to get this automatic, you can run the following command:
`make BOARD=pca10040 FROZEN_MPY_DIR=freeze MICROPY_VFS_LFS2=1`, then it should auto-format the region in case it cannot locate the filesystem.

Also, a note. The `Makefile` in the PR currently defines `FS_SIZE ?= 256K`, and not 64K as documented in the `README.md`. I'll fix this. Anyway, right now you might want to put this a bit lower for `pca10040`. You can easily do this with a extra make parameter setting the size you would like. For example:
`make BOARD=pca10040 FROZEN_MPY_DIR=freeze MICROPY_VFS_LFS2=1 FS_SIZE=64K`.

someusername-123456, 2020-07-15:
Hi Glenn,
Thanks very much for the continued support.

I've compiled with the following parameters as suggested:
`make BOARD=pca10040 FROZEN_MPY_DIR=freeze MICROPY_VFS_LFS2=1 FS_SIZE=64K`
and I've added the 3 lines to `boards/pca10040/mpconfigboard.h`

However, the compilation causes the following issue:
[image]

After a bit of research I set LTO=0 so now the make statement reads as follows:
`make BOARD=pca10040 FROZEN_MPY_DIR=freeze MICROPY_VFS_LFS2=1 FS_SIZE=64K LTO=0`

This causes a different error
[image]

NAshworth2, 2020-10-27:
I'm having the same issue (undefined references to 'disk_ioctl' and 'mp_fat_vfs_type').  Did you resolve this?